Kinovarobotics/ros_kortex #223: Action client not connected: gen3_joint_trajectory_controller/follow_joint_trajectory

Bmoradi93, 2022-03-13:
## Description

Hi KinovaRobotics,
There is a problem with running kortext-driver. launch on Noetic Branch.

I'm getting the following error:

Action client not connected: gen3_joint_trajectory_controller/follow_joint_trajectory

[my_gen3/my_gen3_driver-2] process has died [pid 67481, exit code -6, cmd /home/behnam/git/kinova_ws/devel/lib/kortex_driver/kortex_arm_driver __name:=m
y_gen3_driver __log:=/home/behnam/.ros/log/0b666354-a2fa-11ec-a2e4-9fd7b72872cb/my_gen3-my_gen3_driver-2.log].
log file: /home/behnam/.ros/log/0b666354-a2fa-11ec-a2e4-9fd7b72872cb/my_gen3-my_gen3_driver-2*.log

Please let me know if you have a solution to this problem.

## Version

ROS distribution : 

Branch and commit you are using : ROS-Noetic

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

felixmaisonneuve, 2022-04-12:
Hi @Bmoradi93,

Can you copy the full output of your `roslaunch kortex_driver kortex_driver.launch`?

I will wait a couple of weeks and close this issue if I don't hear from you by then.

Best,
Felix

Bmoradi93, 2022-04-12:
hi @felixmaisonneuve 
Sorry for my late reply,

I solved this issue by upgrading the firmware of the robot.
Thanks

I've also created a docker image for the entire Kinova gen3 ros-packages.

Please let me know if you are interested in having them. I can send a PR.

felixmaisonneuve, 2022-04-18:
Thank you for this offer.

I will kindly decline since we don't really need this feature at the moment. I will keep it in mind if plans change in the future. 

Best regards,
Felix

closed 2022-04-18 by felixmaisonneuve

ArchdukeHans, 2022-06-30:
Hi, @Bmoradi93 

Can you tell me how to update firmware of the robot? I also meet this problem and I cannot resolve it.

Thank you for your time!