Kinovarobotics/ros_kortex #224: Failed requirement 'kortex_api_cpp/2.3.0-r.34@kortex/stable' from 'conanfile.py'

JingKangSun, 2022-03-25:
When I do catkin_make, this error occurs. How can I solve this problem. My environment is: Ubuntu16.04, ros kinetic.

[image]

Thanks!

felixmaisonneuve, 2022-03-25:
Hi @JingKangSun 

For future references, I will link these three other issues related to conan SSL errors : 
#198 / #203 : both are the same error
#210 

In your case, I think you could try
```
conan remote remove conan-center
conan remote add conancenter https://center.conan.io --insert 1
```

This was the solution for the third issue (#210)

Best,
Felix

JingKangSun, 2022-03-26:
Hi, @felixmaisonneuve 

Thanks for your reply, it does work！

Best, 
JingKang

soroush-msd, 2022-03-31:
Hi, when I try the commands I get the following errors: Is there any fix to this?

parallels@ubuntu-linux-20-04-desktop:~/catkin_ws$ conan remote remove conan-center
ERROR: No remote 'conan-center' defined in remotes
parallels@ubuntu-linux-20-04-desktop:~/catkin_ws$ conan remote add conancenter https://center.conan.io --insert 1
ERROR: Remote 'conancenter' already exists in remotes (use update to modify)

soroush-msd, 2022-04-14:
Thank you @felixmaisonneuve .

Using --force tag resolved the issue.