Kinovarobotics/ros_kortex #227: Invoking "make cmake_check_build_system" failed

soroush-msd, 2022-03-30:
Hi, I am trying to run ros_kortex on Ubuntu 20.04.2 ARM64 with ROS Noetic. I have done all the setup and build steps. But when I run catkin_make, I get the following error:

I was wondering if someone could help me address this issue. 
I'd really appreciate your help.

-- Conan: checking conan executable in path
-- Conan: Found program /usr/local/bin/conan
-- Conan: Version found Conan version 1.46.2

-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=9;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /home/parallels/catkin_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=9 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update
Configuration:
[settings]
arch=armv8
arch_build=armv8
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=9
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]

conanfile.py: Installing package
Requirements
    kortex_api_cpp/2.3.0-r.34@kortex/stable from 'kinova_public' - Cache
Packages
    kortex_api_cpp/2.3.0-r.34@kortex/stable:227cd850c3226dee2456c1f7b318da6887458d58 - Missing

Installing (downloading, building) binaries...

ERROR: Missing binary: kortex_api_cpp/2.3.0-r.34@kortex/stable:227cd850c3226dee2456c1f7b318da6887458d58
kortex_api_cpp/2.3.0-r.34@kortex/stable: WARN: Can't find a 'kortex_api_cpp/2.3.0-r.34@kortex/stable' package for the specified settings, options and dependencies:
- Settings: arch=armv8, build_type=Release, compiler=gcc, compiler.libcxx=libstdc++11, compiler.version=5, os=Linux
- Options: shared=False
- Dependencies: 
- Requirements: 
- Package ID: 227cd850c3226dee2456c1f7b318da6887458d58

ERROR: Missing prebuilt package for 'kortex_api_cpp/2.3.0-r.34@kortex/stable'
Use 'conan search kortex_api_cpp/2.3.0-r.34@kortex/stable --table=table.html -r=remote' and open the table.html file to see available packages
Or try to build locally from sources with '--build=kortex_api_cpp'

More Info at 'https://docs.conan.io/en/latest/faq/troubleshooting.html#error-missing-prebuilt-package'

CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:72 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/home/parallels/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/parallels/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:1580: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

felixmaisonneuve, 2022-04-12:
Hi @soroush-msd,

Did you manage to fix your issue?

I will wait a couple of weeks and close this issue if I don't hear from you by then.

Regards,
Felix

soroush-msd, 2022-04-14:
Hi @felixmaisonneuve , thank you for your reply.

I tried to run the command:
`conan remote add conancenter https://center.conan.io --force --insert 1`
as you suggested in the other issue and it seems to have worked.

then I did 
`catkin_make clean` 
`catkin_make`
without specifying the target.

and I got the following errors:

`In file included from /home/parallels/catkin_ws/src/ros_kortex/kortex_driver/src/generated/robot/actuatorconfig_ros_converter.cpp:17:
/home/parallels/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/generated/robot/actuatorconfig_ros_converter.h:28:10: fatal error: ActuatorConfig.pb.h: No such file or directory
   28 | #include <ActuatorConfig.pb.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/parallels/catkin_ws/src/ros_kortex/kortex_driver/src/generated/robot/actuatorconfig_proto_converter.cpp:17:
/home/parallels/catkin_ws/src/ros_kortex/kortex_driver/include/kortex_driver/generated/robot/actuatorconfig_proto_converter.h:28:10: fatal error: ActuatorConfig.pb.h: No such file or directory
   28 | #include <ActuatorConfig.pb.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:76: ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/robot/actuatorconfig_ros_converter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:63: ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/robot/actuatorconfig_proto_converter.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5651: ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
`

after running catkin make again by specifying the target:
` catkin_make --cmake-args -DCONAN_TARGET_PLATFORM=armv8`

I got the following errors:

`Make Error at /opt/ros/noetic/share/catkin/cmake/test/gtest.cmake:180 (add_executable):
  Target "kortex_arm_driver_func_tests" links to target
  "CONAN_PKG::kortex_api_cpp" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/test/gtest.cmake:89 (_catkin_add_executable_with_google_test)
  /opt/ros/noetic/share/catkin/cmake/test/gtest.cmake:37 (_catkin_add_google_test)
  ros_kortex/kortex_driver/CMakeLists.txt:147 (catkin_add_gtest)

CMake Error at /opt/ros/noetic/share/catkin/cmake/test/gtest.cmake:180 (add_executable):
  Target "kortex_arm_driver_func_tests" links to target
  "CONAN_PKG::kortex_api_cpp" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/test/gtest.cmake:89 (_catkin_add_executable_with_google_test)
  /opt/ros/noetic/share/catkin/cmake/test/gtest.cmake:37 (_catkin_add_google_test)
  ros_kortex/kortex_driver/CMakeLists.txt:147 (catkin_add_gtest)

CMake Error at ros_kortex/kortex_driver/CMakeLists.txt:127 (add_library):
  Target "kortex_arm_driver_implementation" links to target
  "CONAN_PKG::kortex_api_cpp" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at ros_kortex/kortex_driver/CMakeLists.txt:127 (add_library):
  Target "kortex_arm_driver_implementation" links to target
  "CONAN_PKG::kortex_api_cpp" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at ros_kortex/kortex_driver/CMakeLists.txt:116 (add_library):
  Target "kortex_driver_generated_files" links to target
  "CONAN_PKG::kortex_api_cpp" but the target was not found.  Perhaps a
  find_package() call is missing for an IMPORTED target, or an ALIAS target
  is missing?

CMake Error at ros_kortex/kortex_driver/CMakeLists.txt:136 (add_executable):
  Target "kortex_arm_driver" links to target "CONAN_PKG::kortex_api_cpp" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at ros_kortex/kortex_driver/CMakeLists.txt:136 (add_executable):
  Target "kortex_arm_driver" links to target "CONAN_PKG::kortex_api_cpp" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at ros_kortex/kortex_driver/CMakeLists.txt:136 (add_executable):
  Target "kortex_arm_driver" links to target "CONAN_PKG::kortex_api_cpp" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
Invoking "cmake" failed`

One a side note, I tried the same setup and installation steps on an Intel based architecture and everything works fine, I guess this might have something to do with Apple m1 chips that I'm running the virtual machine on.

closed 2022-04-18 by soroush-msd