Kinovarobotics/ros_kortex #229: Getting base_command via rostopic

th1991-01, 2022-04-05:
Thanks to `ros_kortex/kortex_driver`, I can subscribe `k_api::BaseCyclic::Feedback base_feedback` via rostopic, like `rostopic echo /my_gen3/base_feedback`. I would also like to do the same for `k_api::BaseCyclic::Command base_command`, but I could not find the corresponding rostopic name. Is it possible to do that via rostopic?

felixmaisonneuve, 2022-04-05:
Hi @th1991-01

Can you please explain a bit more what you are trying to do?

The `k_api::BaseCyclic::Command` is the command you send to you arm when in low level control, but ROS is using high level control, so there is no `k_api::BaseCyclic::Command` in ROS.

Best,
Felix

th1991-01, 2022-04-05:
Sorry for my unclear question.

I can see `base_feedback` via rostopic, even when not controling via ROS, e.g. when controling via Kortex API. I feel that it is very nice. I would like to know if I can also see `base_command` used in Kortex API, in addition to `base_feedback`.

felixmaisonneuve, 2022-04-05:
This not something available and I don't think this is something possible. The API does not provide a call you could use to read the previous command sent in low level servoing, so it is not possible for ROS to know what low level commands were sent from an external program.

Sorry for the inconveniance, 
Felix

th1991-01, 2022-04-05:
I understand. Thank you so much for your kind explanation.

closed 2022-04-05 by th1991-01