Kinovarobotics/ros_kortex #233: Error while gripper is trying to pickup the object

rajnishtripathi10121995, 2022-05-19:
While trying to detect and pick an object using kinova gen3 arm on real robot, I'm getting following error when arm nearly reached pickup location and started trying to pick the object.  

[ INFO] [1652878139.156198293]: Joint Trajectory Goal is accepted.
[ERROR] [1652878139.182840961]: Joint Trajectory failed validation in the arm.
[ERROR] [1652878139.182894755]: Error 1 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (3) and Segment (10).
[ INFO] [1652878139.183348926]: Controller gen3_joint_trajectory_controller successfully finished
[ WARN] [1652878139.183423253]: Controller handle gen3_joint_trajectory_controller reports status ABORTED
Contact information is not stored.
[ INFO] [1652878944.943616215]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1652878944.944051872]: Completed listing of explanations for invalid states.

ROS distribution : ros-melodic

[image]

felixmaisonneuve, 2022-05-19:
Hi @rajnishtripathi10121995,

This is something that could be fixed by changing the time parametrization used when creating waypoint list. 
Time parametrization was changed when moving from Kortex2.2 to [Kortex2.3](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0), so make sure your arm is up to date and you are using a recent commit on you ROS stack. This is probably already the case, because I think you wouldn't get this far otherwise. 

Basically, the real arm cannot go from 0 to max acceleration instantenously, there is an acceleration of acceleration (jerk). Because of that, it can create this kind of errors when generating a waypoint list based on the arm's limits.

I would recommand to go through #206, where we talked quite a bit on this topic. There is also some graphes on this [particular comment](https://github.com/Kinovarobotics/ros_kortex/issues/206#issuecomment-970178986) that can help visualize what is happening. 

My [last comment](https://github.com/Kinovarobotics/ros_kortex/issues/206#issuecomment-970568942) shows how to change the TimeParametrization algorithm. I think that will probably fix your issue. 

Best Regards,
Felix

rajnishtripathi10121995, 2022-05-19:
Hi, my arm is up to date and I'm using a previous commit of ros_kortex because updating ros_kortex leads to several other errors in my existing project.

I have few questions:

I'm using kortex_driver with use_hard_limits as false, should I change it to true to increase acceleration limit of arm? 
If yes, can I set use_hard_limits as true in kortex_driver and use_hard_limits as false in moveit ?

In which configuration file I can change max_acceleration values which are used in generating waypoint?

In following error: "Error 1 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (3) and Segment (10)." What does segment(10) refers to?

felixmaisonneuve, 2022-05-24:
Changing the acceleration limits probably won't fix the issue, I am not even sure if it will reduce the occurence of it. 
If you want to play with them anyway, limits (used by moveit) are defined in yaml files (make sure you change the correct files based on your arm's configuration) : 
[default_joint_limits.yaml](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_move_it_config/gen3_move_it_config/config/7dof/default_joint_limits.yaml) is used when `use_hard_limits` is set to false
[hard_joint_limits.yaml](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_move_it_config/gen3_move_it_config/config/7dof/hard_joint_limits.yaml) is used when `use_hard_limits` is set to true

You arm and moveit could be using different limits, but again, I don't think it will fix the issue and it's not ideal to move slower just because of that.

I would try to change the TimeParametrization (since I am 99% sure this is the issue). 
You shouldn't have to update your ros_kortex repo to do that.  
You only need to change this line to use `AddIterativeSplineParameterization` instead.
https://github.com/Kinovarobotics/ros_kortex/blob/34d4ba416ab1d03b85bebdcbf1983f941eb2f95c/kortex_move_it_config/gen3_move_it_config/launch/ompl_planning_pipeline.launch.xml#L10

You might have to delete `industrial_trajectory_filters/UniformSampleFilter` also, it is not preent in more recent commits and I don't know what it does.

Best,
Felix

rajnishtripathi10121995, 2022-05-27:
Hi @felixmaisonneuve , I have updated my code to recent commit and still I'm getting these errors :

[image]

Invalid Joint Position - Provided joint position for actuator (5) exceeds the joint limits

Joint Trajectory failed validation in the arm.
[ERROR] [1653666006.028418751]: Error 1 : Invalid Speed - Provided Waypoints Lead to go Over Maximum Speed on Actuator (3) and Segment (10).
[ERROR] [1653666006.028753738]: Error 2 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (3) and Segment (10).
 Joint Trajectory failed validation in the arm.
[ERROR] [1653666006.028418751]: Error 1 : Invalid Speed - Provided Waypoints Lead to go Over Maximum Speed on Actuator (3) and Segment (10).
[ERROR] [1653666006.028753738]: Error 2 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (3) and Segment (10).

rajnishtripathi10121995, 2022-05-27:
Is there any other configuration files like default_joint_limits.yaml where I can set max_acceleration/velocity for path planning.

felixmaisonneuve, 2022-06-02:
There is no other configuration files to set max_acceleration/velocity for path planning. 
These are the ones MoveIt uses. 

Just to make sure, are you using MoveIt or Moveit2 to generate your trajectory?

rajnishtripathi10121995, 2022-06-03:
I'm using moveit.
Currently, I have changed all velocity and acceleration limit values in default_joint_limits.yaml with very low values. So, I'm not getting this error. But arm is moving very slow.

felixmaisonneuve, 2022-06-06:
This might reduce the occurence of the error, but I don't think it will fix it entirely, and it makes the arm very painfull to use.
It feels like this error comes from the MoveIt Time Parametrization algorithm. 
Maybe you could try to use a different algorithm : https://ros-planning.github.io/moveit_tutorials/doc/time_parameterization/time_parameterization_tutorial.html

However, the error always happens at the same segment. I would expect the trajectory to have slight differences between each run. Is the trajectory always moving from the same initial position to the same final postion?

rajnishtripathi10121995, 2022-06-07:
I have already tried all three time_parameterization algorithms.
Also, in my case it mostly occurs actuator 3 but sometimes it occurs on other actuators/segments too.

felixmaisonneuve, 2022-06-07:
I don't know, then. 
Since it also happens for other joints and segment, I am almost certain it has to do with MoveIt Time Parametrization algorithm, but I don't know why changing it doesn't work. 

I do not think I can help you more on this,
I am very sorry,
Felix

rajnishtripathi10121995, 2022-06-07:
No issues, thanks for looking into it.

felixmaisonneuve, 2022-11-24:
Hi @rajnishtripathi10121995,

Did you, at any point try to set the arm's velocity and acceleration software limits to the maximum value?
If you have software limits set, the error message would be the same, but I am not sure MoveIt handles software limits, so it might try to reach highier velocties/accelerations and causing errors.

To chnage your software velocity/acceleration limits, you can go in the WebApp, "Configurations"->"Speed Limits". You go to "Advanced" and select the "Cartesian Waypoint Trajectory" (but you might as well set the limits for every trajectory mode just to be safe). Then you click "Set Max" for every limit
[image]

lytridic, 2023-06-08:
Probably a silly question, but here it is: The joint acceleration limits, as in the figure, cannot be set. the SET MAX button does nothing and the APPLY button is greyed out. Should I change the control mode of the robot, and how?

felixmaisonneuve, 2023-06-08:
Hi @lytridic,
Acceleration limits cannot be changed for cartesian waypoints. They are always set to max. 
You can change them for other movement types. 

You have different velocity/acceleration limits depending on the command you use. Moveit uses waypoints, but if you are using another type of command, you need to change the according limits

Best,
Felix

lytridic, 2023-06-09:
Thank you, @felixmaisonneuve 
I had the same problem (invalid speed and invalid acceleration) while attempting to execute a cartesian path using MoveIt, so I tried your solution using the web app, but I still get the same error. The path works fine in the simulation, but not on the real arm. I have the latest version of the ros_kortex package so I don't think time parameterisation is the problem. Next step is to try other limits in the moveit package yaml files

lytridic, 2023-06-09:
At the end of the day, I was finally able to execute the cartesian path by setting all velocities and accelerations from the web app, regardless the trajectory mode. I don't know what did it though. Just setting the max values in only the cartesian waypoints mode did not fix it.

felixmaisonneuve, 2023-06-09:
Curious,

I thought our Joint Controller (used by moveIt) was using waypoints, but maybe I am wrong. I know it used PlayPreCcomputedJointTrajectory before, but that is if you are using an old commit from a couple years ago.

Anyway, solution was to set all velocity and acceleration limits to max via the WebApp apparently. 

@lytridic, do you think we can close this issue?

Best,
Felix

lytridic, 2023-06-09:
Yes I think it can be closed. I hope it helps somebody. It would be nice though if the trajectory mode whose values need to be changed could be determined, instead of changing everything ad hoc. This may have unintended consequences (such as higher speed to poses using the gamepad).

closed 2023-06-16 by felixmaisonneuve

zhangzhiyi22, 2024-06-05:
how  do i deal with the   velocities and accelerations  from the web app,

martinleroux, 2024-06-05:
> how do i deal with the velocities and accelerations from the web app,

Simply navigate to Configurations > Speed Limits.
In the Basic tab, you can set limits that apply in every mode. In the Advanced tab, you can set different limits for different modes of control.

zhangzhiyi22, 2024-06-05:
but the  model  i used  is gen3_lite,it  seems that  nothing happens.

martinleroux, 2024-06-05:
The interface is the same for Gen3 lite, the only difference is that Gen3 lite's max limits are much lower than Gen3's. Since this issue is closed, I recommend contacting us at support@kinova.ca is you need further assistance.

zhangzhiyi22, 2024-06-05:
Action client not connected: gluon_controller/follow_joint_trajectory    this is the error  for  gen3  when roslaunch kortex_driver kortex_driver.launch.How to deal with if?

smoya23, 2024-06-05:
@zhangzhiyi22 Could you open a separate issue with details on the steps you followed to get to this issue (and screenshots if possible) ?