Kinovarobotics/ros_kortex #234: Error - Unable to load type while running rosservice info

manishgupta1208, 2022-05-25:
I am new to ROS, currently running ROS Kortex on Ubuntu for simulation with Gazebo.
I followed the instructions for installation. 

After installation, I run the command for spawning Gen3 robot - roslaunch kortex_gazebo spawn_kortex_robot.launch
Everything runs fine. I tried examples from python to send robot to home. Those work fine too.

Next, we were trying to connect through Matlab ROS toolbox to connect to ROS network. 
Subscribing to joint_topics is fine and latest joint values can be fetched. 

Now issue is in connecting to ros-services and there is error "Unrecognized message type in KortexDriver...." 
For checking, we tried to get info for this service directly using rosservice info and error received is Unable to load type [KortexDriver]

Replicate Error:

1. First terminal source devel and then run this command roslaunch kortex_gazebo spawn_kortex_robot.launch
2. In another terminal run this command "rosservice info /my/gen3/base/read_action"

I tried commands for other services too but getting similar errors.

[image]

felixmaisonneuve, 2022-05-25:
HI @manishgupta1208,

This is because not all services are implemented in a Gazebo simulation. 
You can find the list of the API calls that are implemented in our simulator in the [kortex_example readme](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros) (not very intuitive, I know)

You will only be able to use services from that list.

Best,
Felix

felixmaisonneuve, 2022-05-25:
OK, I think I nuderstand your problem. 
We use custom service messages that are probably not recognized by the MatLab ROS toolbox. 
All our services and messages are defined [here](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_driver/srv) and [here](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_driver/msg).

There is surely a way to link or copy them somewhere so your MatLab ROS toolbox recognizes the messages types. I don't if you have something similar in MatLab, but in our ROS repo, we add everything in the kortex_driver CMakeLists.txt file : 
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_driver/CMakeLists.txt#L46-L55

Best,
Felix

manishgupta1208, 2022-05-26:
Thanks for sharing the suggestions Felix. I totally agree that Matlab is not able to understand the custom services messages type and will try to find a way to link with Matlab.

Thanks,
Manish

felixmaisonneuve, 2022-05-26:
I will close this issue since this seems to be the issue, but don't hesitate to re-open it if you have any question or if it doesn't resolve it.

Best,
Felix