Kinovarobotics/ros_kortex #236: Force control mode in ROS (Kinova Gen3)

AlirezaMsb, 2022-05-30:
Hi, 

I have a few technical questions regarding Gen3 arms and I would appreciate it if you could help me with them, mainly I am wondering about the "Force control" mode (not admittance mode) in ROS:

1. In the manual "force control" mode is mentioned as a part of the API. Does the same function exist in the ROS pkg as well? I found [SendWrenchCommand](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/srv/generated/base/SendWrenchCommand.srv) ROS service, I am wondering whether this does what I am looking for.

2. Does the ROS pkg publish e.g. a wrench topic representing the external wrench applied to the end-effector (That is to say the torque sensor values moved to the end-effector)? 

I need this information for developing a special kind of force control.
I look forward to hearing from you,

Alireza

felixmaisonneuve, 2022-06-02:
Hi @AlirezaMsb,

The ROS packages is generated automatically from the API, so yes the function exist in ROS. However, even though the `SendWrenchCommand` call is available through the API, the actual function is not implemented in the arm and it will result in an error. So, even if it exists, you cannot use `SendWrenchCommand`.

The external wrench applied to the end effector is published on the `base_feedback` ROS topic. You should find the Wrench on `base_feedback->base->tool_external_wrench_force_x/y/z` and  `base_feedback->base->tool_external_wrench_torque_x/y/z`

If you want to implement Force Control, you will have to continuously monitor this topic and react based on the value you get. 

Best,
Felix

AlirezaMsb, 2022-06-06:
@felixmaisonneuve 

Hey Felix, 
Thanks for your help,
So if my understanding is correct, "force control" mode is neither available through the API nor ROS (as a result). 
Please correct me if I am wrong. 
I think that would be the answer to my question, so I'll close the issue. 
I appreciate your help,

Alireza

closed 2022-06-06 by AlirezaMsb

felixmaisonneuve, 2022-06-07:
@AlirezaMsb, this is correct

Best of luck,
Felix