Kinovarobotics/ros_kortex #238: Gripper closing issue in Gen3 Lite

GitNzino, 2022-06-15:
## Description

When I try to execute the command `example.example_send_gripper_command(gripper_close)` with `gipper_close = 1.0` the gripper only close to 50%

## Version

ROS distribution : melodic

Branch and commit you are using : latest

## Other Curiosity

It is possible to deactivate the protection via ros python as showed in the web app?
[image]

Thank you in advice and have a good day,
Lorenzo

GitNzino, 2022-06-16:
Hi,
thank you for response.

The gripper problem was about the time to reach the goal pose.

For the protection zone i added this function in example code code:

```python
def protection_zone_mod(self,radius):
        protection_zone_service = '/' + self.robot_name + '/base/update_protection_zone'
        rospy.wait_for_service(protection_zone_service)
        self.protection_zone_service = rospy.ServiceProxy(protection_zone_service, ProtectionZone)
        protection_zone = ProtectionZone()
        protection_zone.shape.dimensions = [radius, 0.35]
        protection_zone.is_enabled = False
        print(protection_zone)
        self.protection_zone_service(protection_zone)
```

and this output:
```bash
process[my_gen3_lite/tablet_movement-1]: started with pid [9141]
[INFO] [1655394495.276525]: Using robot_name my_gen3_lite , robot has 6 degrees of freedom and is_gripper_present is True
Traceback (most recent call last):
  File "/home/lorenzo/cw_kinova/src/kinova/kortex_movement/scripts/tablet_movement.py", line 543, in <module>
    main()
  File "/home/lorenzo/cw_kinova/src/kinova/kortex_movement/scripts/tablet_movement.py", line 526, in main
    example.protection_zone_mod(0.08)
  File "/home/lorenzo/cw_kinova/src/kinova/kortex_movement/scripts/tablet_movement.py", line 124, in protection_zone_mod
    self.protection_zone_service = rospy.ServiceProxy(protection_zone_service, ProtectionZone)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 411, in __init__
    super(ServiceProxy, self).__init__(name, service_class)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/service.py", line 59, in __init__
    self.request_class = service_class._request_class
AttributeError: type object 'ProtectionZone' has no attribute '_request_class'
[my_gen3_lite/tablet_movement-1] process has died [pid 9141, exit code 1, cmd /home/lorenzo/cw_kinova/src/kinova/kortex_movement/scripts/tablet_movement.py __name:=tablet_movement __log:=/home/lorenzo/.ros/log/76c7a370-ed8b-11ec-8fb0-3c950922ab5a/my_gen3_lite-tablet_movement-1.log].
log file: /home/lorenzo/.ros/log/76c7a370-ed8b-11ec-8fb0-3c950922ab5a/my_gen3_lite-tablet_movement-1*.log

```

What could I do?

felixmaisonneuve, 2022-06-16:
The `protection_zone_service` takes a request. Looking at other examples in the repo, I feel like you should do something like 
```python
from kortex_driver.srv import ProtectionZone, ProtectionZoneRequest # this is probably import * in your case so you don't have to worry about it

def protection_zone_mod(self,radius):
        protection_zone_service = '/' + self.robot_name + '/base/update_protection_zone'
        rospy.wait_for_service(protection_zone_service)
        self.protection_zone_service = rospy.ServiceProxy(protection_zone_service, ProtectionZone)
        protection_zone_request = ProtectionZoneRequest()
        protection_zone_request.input.shape.dimensions = [radius, 0.35]
        protection_zone_request.input.is_enabled = False
        print(protection_zone_request)
        self.protection_zone_service(protection_zone_request)
```

But even with that, I do not think it will update your protection zone because you probably need to specify the ProtectionZone `handle` so the arm knows which protection zone you are trying to update. This is why you might need to ReadAllProtectionZones, to find the handle of the protection zone your are trying to edit.

I cannot test it at the moment, if it doesn't work and you still have issues, I will be glad to help you
Best,
Felix

GitNzino, 2022-06-17:
Hi,

Thank you very much for support

I tried your code but I notice this issues:

- The `ProtectionZoneRequest()` does not exist
- Trying to do the same things with other simila class like: `UpdateProtectionZone()` i cannot obtain the rigth interaction

What should i do?

Thank you in advice

GitNzino, 2022-06-20:
Hi,

At least I solved in this manner:

```python
        protection_zone_reader = '/' + self. robot_name + '/base/read_all_protection_zones'
        rospy.wait_for_service(protection_zone_reader)
        self.protection_zone_reader = rospy.ServiceProxy(protection_zone_reader, ReadAllProtectionZones)
        protection_zone_read = ReadAllProtectionZonesRequest()
        
        zone = self.protection_zone_reader(protection_zone_read)
        
        protection_zone_service = '/' + self.robot_name + '/base/update_protection_zone'
        rospy.wait_for_service(protection_zone_service)
        self.protection_zone_service = rospy.ServiceProxy(protection_zone_service, UpdateProtectionZone)
        
        send = UpdateProtectionZoneRequest(zone.output.protection_zones[0])     
        
        send.input.shape.dimensions = [0.35, radius]
        send.input.is_enabled = True

        self.protection_zone_service(send)   
```     
Do you think is it right?

felixmaisonneuve, 2022-06-20:
Yes this is actually what it should look like.

Does this work? 
When changing `send.input.is_enabled = True` to `send.input.is_enabled = False`, does the protection de-activate in the web app?

GitNzino, 2022-06-20:
Yes, it does!
Also when the area is resized also in the web app appears.

closed 2022-06-20 by felixmaisonneuve