Kinovarobotics/ros_kortex #239: Force control for various arm base configuration

AlirezaMsb, 2022-06-16:
Hi,

We are currently using two gen3 6dofs arms in a robot, and since the robot bases are not located on the horizontal plane (the arms are almost upside down), the admittance mode does not work properly as expected due to the change in the gravity terms. Is there any e.g. rosservice for setting the orientation of the arm base in that regard? 
Also I noticed something strange in the arm that we are using. When I was monitoring the external wrench topic on one of the arms, the force value sometimes changed slowly overtime (sometimes even around 10-20N), although there was no external force/torque being applied to the arm. Can you think of anything that might cause this problem?

I appreciate your help,

Alireza

felixmaisonneuve, 2022-06-16:
Hi @AlirezaMsb,

You need to change the gravity vector with the API call `SetGravityVector`. You will find the documentation for this function in [ControlConfig](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/ControlConfig.md). You will need to specify your specific [GravityVector](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/ControlConfig/GravityVector.md#).

You can access this from ROS by calling our auto-generated service : 
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_driver/src/generated/robot/controlconfig_services.cpp#L61

Your external wrench changing overtime is very strange. This might be because your arm is not mounted normally and your gravity vector is not updated to reflect that. It could also be a could a good idea to reset the torques in your arm. The best to do it is by putting your arm in a vertical position to reduce cross torque from the gravity. Any other configuration is not optimal and is not recommanded. Since your arm is almost upside down, I would say a reset torque shouldn't be too bad with your arm pointing a the ground. With only a small cross torque on the second joint, I would say it shouldn't be that big of a deal.  
If you still experience the issue after changing the gravity vector and reseting the torque values, please let me know and we will take a better look at it. 

Regards,
Felix

AlirezaMsb, 2022-06-19:
Thanks for your reply @felixmaisonneuve ,

I understand. I assume by calling the service/using the API the torque/wrench values at the end-effector will be corrected too. Please correct me if I am wrong. 
Regarding the wrench values at the end-effector, from #164 apparently, the published value includes the gravity terms as well. I believe in the literature usually this term is separated from the external wrenches coming from other sources. Just to confirm, has there been any updates in this regard in the repository since the last discussion in #164 ? That is to say, are the wrench values without the gravity terms accessible?

Thanks for your help Felix,

Alireza

felixmaisonneuve, 2022-06-20:
I went and tried the external_tool_wrench and the gravity force is not shown in the values (meaning everything should be close to 0 regardless of the position). Values will change only if you push/twist the arm. 

It is important to note that the values are computed based on the torque values of every joint and it does not come for an actual sensor. This means it is (quite) imprecise. Resetting torque sensors will help a bit, but even then values won't be very precise. 

I encourage you to test it yourself. Un-official tests were done at Kinova and results showed a ~8% error for static poses (no tests were done in motion). This will probably vary a lot from one arm to another. 

I will edit my comment in the other issue. 

Best,
Felix

AlirezaMsb, 2022-06-21:
I understand. 
The approximate error data coming from unofficial tests at Kinova was a very valuable information. 
Thanks for your help @felixmaisonneuve , I will close this issue.

Alireza

closed 2022-06-21 by AlirezaMsb