Kinovarobotics/ros_kortex #240: Is it avaliable to achive torque control of  Gen3 with a more simplify method?

ChanSir-QZ, 2022-06-21:
## Summary

Define an object for the Gen3 to handle the control signals and read the sensors. Using the method robot.control() you read the joint values at 1kHz.
Refer to https://github.com/cpezzato/active_inference/blob/master/AICpanda.cpp

## Use case

Why is this feature needed?

Torque control could be used for different kinds of optimazation.

ChanSir-QZ, 2022-06-29:
Hi @felixmaisonneuve ,
I have tried the example on how to use the torque control in kortex repo. It is ok.
And i was wandering if i could control the robotic arm based torque control using ROS. Sorry for mine pool knowledge and proposing an implicit issue.

felixmaisonneuve, 2022-06-29:
No, it is not possible.

Unfortunately, the torque control mode is only available in low level servoing mode, which is not available in our ROS stack. The only way to use the torque control mode is by using the API directly. 

Best,
Felix

ChanSir-QZ, 2022-06-30:
> 

Thank you for your reply. 

Kind regard,

Chen

closed 2022-06-30 by ChanSir-QZ