Kinovarobotics/ros_kortex #242: Sending velocity twist command from the terminal

RaduCorcodel, 2022-06-25:
Hello,
I'm trying to send a twist command to the robot using ```rostopic pub /my_gen3/in/cartesian_velocity kortex_driver/TwistCommand 'reference_frame:0, twist:{linear_x:0.0, linear_y:0.0, linear_z:0.05, angular_x:0.0, angular_y:0.0, angular_z:0.0}, duration:0'``` and I get back 

```
ERROR: Not enough arguments:
 * Given: ['reference_frame:0, twist:{linear_x:0.0, linear_y:0.0, linear_z:0.05, angular_x:0.0, angular_y:0.0, angular_z:0.0}, duration:0']
 * Expected: ['reference_frame', 'twist', 'duration']

Args are: [reference_frame twist.linear_x twist.linear_y twist.linear_z twist.angular_x twist.angular_y twist.angular_z duration]
```

It must be something minor because if I do ```rostopic pub /my_gen3/in/cartesian_velocity kortex_driver/TwistCommand 'reference_frame: 0, twist:{linear_x:0.0, linear_y:0.0, linear_z:0.05, angular_x:0.0, angular_y:0.0, angular_z:0.0}, duration:0'``` (notice the additional space between "reference_frame:"  and "0", I get back ```[WARN] [1656174104.873976]: Inbound TCP/IP connection failed: field reference_frame must be unsigned integer type```.

Ca you please give me an example of a simple velocity command I can use? The one in the examples doesn't work either.  

For reference, I'm on ROS Noetic.

Thank you,
Radu

RaduCorcodel, 2022-06-25:
Alright, I sorted it out myself. It appears ROS is very finicky about spaces but here's what worked for me on **Noetic**:

```
rostopic pub /my_gen3/in/cartesian_velocity kortex_driver/TwistCommand '{reference_frame: 0, twist: {linear_x: 0.0, linear_y: 0.0, linear_z: 0.01, angular_x: 0.0, angular_y: 0.0, angular_z: 0.0}, duration: 0}'
```
Also note there's another curly brace set at the beginning and end of the message.

Hope it helps anyone who had this issue.

Best,
Radu

felixmaisonneuve, 2022-06-29:
Hi @RaduCorcodel,

I am glad you managed to sort it out. 

Sending commands via `rostopic` can sometimes be a pain because ros is very picky with its format and it gets difficult with more "complex" data structures. To prevent this, I use the "Tab" key (on the keyboard) and it is usually able to autocomplete the topic name, the data type and the data format, then I only have to go back and change the values that interest me. 

Regards,
Felix

closed 2022-06-29 by felixmaisonneuve

RaduCorcodel, 2022-07-11:
Hi @felixmaisonneuve,

Sorry to add to this but I just needed one more tip. I'm trying to send a velocity command along the X-axis of the end-effector (which is not aligned with the X-axis of the base). Your tip to autocomplete with the tab key works great across different ROS distros but I can't get it to execute the velocity command in the EF's frame. I tried changing the 'reference_frame' from 0 to 6 but there's no effect; the command is still in the frame of the base. Am I missing something?
Thanks again!

Best,
Radu