Kinovarobotics/ros_kortex #243: Is there a gravity compensated mode api like kinova gen2 in kotex?

lexidliny, 2022-07-11:
Hello
I am using kinova gen3 kotex with 7 freedom recently.  I have two questions to be solved.
Now I am going to realize a force control method through torque control. I found that the arm will fall when the torques of joints are set to be zero. So I think it because the effect of gravity. Then I want to found some api that used to compenste the gravity, but there seems to be nothing. 
1. Is there a gravity compensated mode api like kinova gen2 in kotex?  
2. How to ensure the real-time rate of torque control at 1khz? In my experiment, it only arrives at about 500 hz.

felixmaisonneuve, 2022-07-12:
Hi @lexidliny,

Low-level control is not possible in ROS. You won't be able to achieve the 1kHz frequency (or maybe it is and I simply don't know how). I usually tell people to use the API directly if they need to use the torque control mode, because it requires low-level control and it does not work in ROS.

Also, there is no gravity compensated mode in our Kortex API. If you want an admittance mode, you will need to implement your own that compensate for gravity in low level. 

Best,
Felix

felixmaisonneuve, 2022-07-12:
Actually, there is the high level `SetAdmittance` function that lets you set in the arm in one of our [AdmittanceMode](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/enums/Base/AdmittanceMode.md#). 
I think this might be what you are looking for.

lexidliny, 2022-07-13:
> 
Sorry, I shoud submit this question in Kortex repo. Actually, I tested my algorithm using the Kortex Ap [01-torque_control_cyclic.cpp](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp). The sending frequency of the torque command is maintained at about 500Hz.

Also as you say, I have made a kortex Gen3 model using [rbdl](https://github.com/rbdl/rbdl) c++ Library to compute the torque caused by gravity.  Unfortunately, I found there is some Kinematics and dynamics error between the model  and real arm. The parameters of the model comes from User Guide.pdf.  

For example, when I input the joint angles which is also the home of real arm like this:
```
    Q[0] = 359.998 / 180 * M_PI;
    Q[1] = 15.004 / 180 * M_PI;
    Q[2] = 180.008 / 180 * M_PI;
    Q[3] = 229.999 / 180 * M_PI;
    Q[4] = 359.999 / 180 * M_PI;
    Q[5] = 55.0 / 180 * M_PI;
    Q[6] = 90.002 / 180 * M_PI;
```
the real arm will go to xyz coordinate `0.46, 0.008, 0.438`, but the result computed by the parameters in user guide is `0.456594 0.000957351    0.433764`.

As for the torque, there is about 1 N.m error between the real arm and the computed result. 

I want to implement a torque algorithm that need to compensate the gravity in low level comtrol mode. Now, It seems to be difficult because of the error. Is there a more precise model of arm? Or  how can I solve the problem?

felixmaisonneuve, 2022-07-13:
Make sure your computer is not running anything in the background when running your program. Having a web browser and other applications running can slow your program too much and prevent you from sending a command every ms. Sometimes, using Ubuntu instead of Windows can make a difference. Also, you could try to optimize the algorithm you are using. 1kHz is very fast and it can be hard for the computer to keep up.

Depending on the age of the arm and its wear, it is possible it has lost a bit of precision. Also, each arm is unique. Shells are imperfect so the position may vary very slightly from one arm to another. I think a 8mm error is not so bad. How precise do you need the arm to be?

Same goes for torque values. They can offset themselves after some times. You can put the arm in a vertical position (all joints to 0), so there is no cross torque generated by gravity. In that position, you can "reset all torques" (possible to do from the webapp). It will reset the torque offset and could help you get more precise results. Since you want to implement a torque algorithm, you could try this first.

Best,
Felix

felixmaisonneuve, 2022-08-15:
Hi @lexidliny,

Can we close this issue?
I will do it anyway in a couple weeks if I do not ear from you. 

Regards,
Felix

closed 2022-11-21 by felixmaisonneuve