Kinovarobotics/ros_kortex #245: Invalid Joint Position - Provided joint position for actuator (5) exceeds the joint limits

rajnishtripathi10121995, 2022-07-21:
While trying to detect and pick an object using kinova gen3 7 DoF arm, I'm getting following error in the middle of trajectory:

```
[ERROR] [1658379235.270373564]: Joint Trajectory failed validation in the arm.
[ERROR] [1658379235.271291979]: Error 1 : Invalid Joint Position - Provided joint position for actuator (5) exceeds the joint limits
```

[image]

I'm getting above error on kortex_driver node and I'm running moveit separately on another system.

ROS distribution : ros-melodic

felixmaisonneuve, 2022-07-21:
Hi @rajnishtripathi10121995,

Visually, is your joint 6 exceeding its joint limit?
Did you play with any config file? Usually, joint limits are specified in the xacro file (used to generate the URDF) and I would expect MoveIt to consider these limits when creating a trajectory, so an error like this should not happen. 
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_description/arms/gen3/7dof/urdf/gen3_macro.xacro#L239

You could add some lines un your joint_trajectory_action_server.cpp and compile it again with `catkin_make` to be have more details about the error : 
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp#L162-L173
```
            for (unsigned int i = 0; i < report.trajectory_error_report().trajectory_error_elements_size(); i++)
            {
                ROS_ERROR("Error %i : %s", i+1, report.trajectory_error_report().trajectory_error_elements(i).message().c_str());
                ROS_ERROR("Error %i : error_type     : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).error_type());
                ROS_ERROR("Error %i : error_value    : %f", i+1, report.trajectory_error_report().trajectory_error_elements(i).error_value());
                ROS_ERROR("Error %i : min_value      : %f", i+1, report.trajectory_error_report().trajectory_error_elements(i).min_value());
                ROS_ERROR("Error %i : max_value      : %f", i+1, report.trajectory_error_report().trajectory_error_elements(i).max_value());
                ROS_ERROR("Error %i : index          : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).index());
                ROS_ERROR("Error %i : waypoint_index : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).waypoint_index());
            }
```

Do you have any other info that could help diagnose this issue?

Regards,
Felix

rajnishtripathi10121995, 2022-07-25:
Hi @felixmaisonneuve,

In config files, I have only reduced max_velocity and max_acceleration in "ros_kortex/kortex_move_it_config/gen3_robotiq_2f_85_move_it_config/config/7dof
/default_joint_limits.yaml".

This error occurs randomly it doesn't reproduced again yet, visually joint6 is not exceeding its limit. 

I was not getting this error earlier, only change that I have done is running MoveIt on another system.

felixmaisonneuve, 2022-07-25:
Do you have protection zones set? 
I am not sure how this could happen since joint limits are not configurable in the arm. 

Also, I don't know if reducing velocity and acceleration will prevent the actuator from going outside its position limit.
It could be interesting to see what is the trajectory sent by moveit and if one of the point gets close the joint limit.

You could try to put your arm in a vertical position as well (all joints to 0). Maybe the actuator has an offset and it leads to this error. If it is the case, you will notice it right away as your arm won't be straight (with all joints at 0).

rajnishtripathi10121995, 2022-07-25:
I haven't configured anything related to protection zones.
Will put arm in vertical position and update you. Might be an actuator offset issue because sometimes object grasping pose while pickup also deviates from its correct pose for grasping.

felixmaisonneuve, 2022-08-15:
Hi @rajnishtripathi10121995 

Was it an actuator offset?
Do you have any feedback?
Can we close this issue?

Regards,
Felix

rajnishtripathi10121995, 2022-08-22:
Hi,
This issue occurs randomly and it was not related to offset in my case.
It haven't occurred from last few weeks, closing this issue for now.
Thanks for looking into this.

closed 2022-08-22 by rajnishtripathi10121995