Kinovarobotics/ros_kortex #249: MoveIT

LeeJeongWoon-Workout, 2022-08-08:
hello 

---setting----
OS: ubuntu 20.04
ROS Noetic

I tried to make my own MoveIt Configuration files with gen3_lite model. 

I loaded gen3_lite_macro.xacro URDF file into MoveIt setupt Assistant but got an error as below
[image]

I knew from README that you made move_if_config with Moveit . but that URDF model not be loaded. You used your own SRDF file for making move_it_config package? If then, Can I get that SRDF file if not illegal?

felixmaisonneuve, 2022-08-08:
Hi @LeeJeongWoon-Workout,

I am not sure why the Setup Assistant is unable to locate the kortex_description package. 
Every arm's configuration as a URDF file in the [kortex_description folder](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_description). You can find the [Gen3-lite URDF](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_description/arms/gen3_lite/6dof/urdf) in there.
You can also generate the urdf from the xacro files using something like 
```
rosrun xacro xacro.py gen3_lite_macro.xacro > gen3_lite.urdf
```

I did not play with the Moveit Setup Assistant that much so I am not sure what the issue could be. 
Do you need to configure it to be in the workspace? Do you need to build with `catkin_make` first? Maybe source devel/setup.bash first before launching the setup assistant?

Best,
Felix

LeeJeongWoon-Workout, 2022-08-08:
I solved my issue. I should have got into kortex_description/robot directory

felixmaisonneuve, 2022-08-09:
Great

I will close this issue then
If you have any other question or problem, feel free to re-open the issue or open a new one

Best Regards,
Felix

closed 2022-08-09 by felixmaisonneuve