Kinovarobotics/ros_kortex #253: kortex_dual_driver.launch does not work

psalidoiri, 2022-09-09:
## Description

When we run the following command **'roslaunch kortex_driver kortex_dual_driver.launch left_ip_address:=192.168.1.12 right_ip_address:=192.168.1.10 '** we do not receive any errors but the Kinova robots do not appear in their real position in RVIZ (attached image).

## Version

ROS distribution : Melodic

Branch and commit you are using : melodic-devel 

[image]

felixmaisonneuve, 2022-09-12:
HI @psalidoiri,

This is expected behaviour. 
By launching `kortex_dual_driver.launch`, your are using the kortex_dual_robots.xacro description
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_driver/launch/kortex_dual_driver.launch#L63
This description loads two Gen3 arms at these positions : 
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_description/multiple_robots/kortex_dual_robots.xacro#L21-L33

The dual arm launch file does not launch the kortex_driver node and does not launch the moveit for this dual arm config.
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_driver/launch/kortex_driver.launch#L60-L97

The first thing you need to do is create your own moveit config with your two arms. That is specific for everyone, it depends on what arms you are using. It is possible to use the MoveIt SetupAssistant (that was used to create every moveit config initially), but I think there was some modifications on the configs with time and the SetupAssistant is not working anymore.This is something that I hope could be fixed in the future. The other option would be to copy one of the config in [kortex_move_it_config](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_move_it_config) and modify each file to match your config, starting with move_group.launch (the entry point used by the kortex_driver.launch)
https://github.com/Kinovarobotics/ros_kortex/blob/7ba75b3e6293fe4af9fe64f35e877d1ff2613360/kortex_driver/launch/kortex_driver.launch#L87

Once this is done, if you want to use the driver (to send commands to the arms), you might need to to modify the update the actual kortex_driver to be able to handle multiple arms at the same time. This is probably A LOT of work, but I never really got the time to look into it. If you want to do something like this, the easiest/fastest way is simply to launch two driver at the same time (unless this causes other conflict issues, I never tried that).

Hope it clear things out for you.
I am open to answer any wuestion you might have on this topic.

Best regards,
Felix

psalidoiri, 2022-09-20:
Hi @[felixmaisonneuve](https://github.com/felixmaisonneuve) ,
I think there is a mistake in your answer. The **dual arm** launch file is launching the **kortex_arm_driver** and this launch file calls the **kortex_driver** launch file:
https://github.com/Kinovarobotics/ros_kortex/blob/cec7161c1e60f1954ba9fc0867377279350b103a/kortex_driver/launch/kortex_dual_driver.launch#L98
https://github.com/Kinovarobotics/ros_kortex/blob/cec7161c1e60f1954ba9fc0867377279350b103a/kortex_driver/launch/kortex_arm_driver.launch#L35

Anyway, these days we have been testing code changes to make the dual_kortex launch file work:

1. We modify the **kortex_arm_driver.launch**:
https://github.com/Kinovarobotics/ros_kortex/blob/cec7161c1e60f1954ba9fc0867377279350b103a/kortex_driver/launch/kortex_arm_driver.launch#L35
```
<node name="$(arg robot_name)_driver" pkg="kortex_driver" type="kortex_arm_driver" output="screen" ns="$(arg robot_name)"> launch-prefix="gdb -ex run args"
```

2. **kortex_description/arms/gen3/7dof/config/joint_limits.yaml**:
https://github.com/Kinovarobotics/ros_kortex/blob/cec7161c1e60f1954ba9fc0867377279350b103a/kortex_description/arms/gen3/7dof/config/joint_limits.yaml#L1-L8
```
 joint_names: 
   - joint_1 
   - joint_2 
   - joint_3 
   - joint_4 
   - joint_5 
   - joint_6 
   - joint_7 
```

3. The config file of the gripper that we are using **kortex_description/grippers/robotiq_2f_85/config/joint_limits.yaml**:
https://github.com/Kinovarobotics/ros_kortex/blob/cec7161c1e60f1954ba9fc0867377279350b103a/kortex_description/grippers/robotiq_2f_85/config/joint_limits.yaml#L1-L2
```
 gripper_joint_names: 
   - finger_joint 
```

4. We modify the **kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp** file, reversing the changes before [this commit](https://github.com/Kinovarobotics/ros_kortex/commit/cc70970bc5a7718541925d381b9e199f2aac77ac) (changes in line 655 and 666):
https://github.com/Kinovarobotics/ros_kortex/blob/cec7161c1e60f1954ba9fc0867377279350b103a/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L655-L666
```
     joint_state.name[i] = m_prefix + m_arm_joint_names[i]; 
     joint_state.position[i] = m_math_util.wrapRadiansFromMinusPiToPi(m_math_util.toRad(base_feedback.actuators[i].position)); 
     joint_state.velocity[i] = m_math_util.toRad(base_feedback.actuators[i].velocity); 
     joint_state.effort[i] = base_feedback.actuators[i].torque; 
 } 
  
 if (isGripperPresent()) 
 { 
     for (int i = 0; i < base_feedback.interconnect.oneof_tool_feedback.gripper_feedback[0].motor.size(); i++) 
     { 
         int joint_state_index = base_feedback.actuators.size() + i; 
         joint_state.name[joint_state_index] = m_prefix + m_gripper_joint_names[i];
```

Now, we can see the Kinova robots in their real position:
`roslaunch kortex_driver kortex_dual_driver.launch left_ip_address:=192.168.1.12 left_gripper:=robotiq_2f_85`

[image]

Best regards,
Pablo S.

felixmaisonneuve, 2022-09-20:
You are right, the dual arm launch file is launching the kortex_arm_driver. 

This is amazing you made it work. Thank you for sharing your work here. 

The commit you mentionned is very interesting. I remember doing this modification because it was causing an error. 
I will keep in mind you managed to make the dual arm config work, but I won't revert the changes in the yaml yet.

Do you need more support on this topic?

Regards,
Felix

psalidoiri, 2022-09-22:
Hi @[felixmaisonneuve](https://github.com/felixmaisonneuve) ,
It would be interesting to apply these changes to a stable branch and have a launch file in which both robots work. People who need to work or experiment with two Kinova robots would benefit from this improvement. For example, in our research center several people work with both robotic arms at the same time and it would be difficult to apply the modifications in each of their computers.

I wish this would happen soon,
Pablo S.

felixmaisonneuve, 2022-09-22:
I created the bugfix/dual-arm-melodic branch that you can checkout. 
Verify that the changes I made are correct and that it works. I have not tested it since I do not have a dual arm setup at the moment, but since you want this fix as soon as possible, I created the branch anyway based on the modifications you described in your previous comment.

Please tell me if I missed something or if this is not what you are looking for

psalidoiri, 2022-09-23:
I just tested the changes in this new branch, and it works as desired.

Thanks, 
Pablo S.

closed 2024-10-16 by aalmrad