Kinovarobotics/ros_kortex #254: end_effector_link.stl not valid

Cfather, 2022-09-20:
## Description

ros_kortex-noetic-devel/kortex_description/arms/gen3/7dof/meshes/end_effector_link.STL is not a valid file. I can not open that.

## Version

ROS distribution : noetic

Branch and commit you are using : ros_kortex-noetic-devel

felixmaisonneuve, 2022-09-20:
Hi @Cfather,

What program are you trying to open the file with?
STL files are 3D objects that can only be opened with special softwares (SolidWorks for example).
Those are binairies that cannot be read from a simple text editor

Best,
Felix

Cfather, 2022-09-20:
Hi, Felix,

Of course, I know STL files can not be opened from a text editor.
So the Matlab R2021b (**importrobot** function in robotics toolbox) in my Ubuntu 20.04 can not recognize this single STL file. It can load the STL files of all other links except for the end effector.
So I tried **meshlab** in Ubuntu then. Again, it can show the STL files of all other links except for the end effector.
Then I tried **3D viewer** in my Windows 10. The result was still the same.
As a result, I was wondering if you uploaded that STL file by mistake. I primarily work in Matlab, so whether the file can be opened by SolidWorks or not does not matter to me. And I think it should be an easy fix because I don't have any problems accessing the rest of the STL files, but only that one (end_effector_link.STL). Would you mind double-checking that STL file for me?

Many thanks,
Bohao Zhang

felixmaisonneuve, 2022-09-21:
You are absolutely right. I tried it and it is not working. Looking at the size of the file, it is only 80 bytes, so it makes sense.
I am not sure what this file is supposed to be. 

I see it is not present for 6dof arm meshes.
I searched the repo and this file is never used anywhere.
I will remove it if that's okay with you. 

Regards,
Felix

Cfather, 2022-09-21:
Hi, Felix,

Thanks for confirming this.
So what should the actual STL file for the end effector be? Or this is just a placeholder and the users can specify any end effectors they want? I am a little bit confused.

Regards,
Bohao

felixmaisonneuve, 2022-09-22:
our end-effector model can be found in the [grippers](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_description/grippers) folder. I am not sure what this file was for originally. Maybe it was a placeholder or it simply is a mistake. 
In any case, it is not working and I don't see a reason to keep it.

Cfather, 2022-09-22:
Thanks for answering my question, Felix. Appreciate it!

PR #261 (2022-10-31): removed corrupted file and clarify readme
Fixes #254 
Removes an unused and corrupted STL file

Clarifies readme so users clone the correct branch for their ROS version
PR #262 (2022-10-31): removed corrupted file and clarify readme
Fixes #254 
Removes an unused and corrupted STL file

Clarifies readme so users clone the correct branch for their ROS version

Add `req` in `set_cartesian_reference_frame()` for the example_full_arm_movement.py

Same as #261 and #257, but for the melodic branch
closed 2022-10-31 by felixmaisonneuve