Kinovarobotics/ros_kortex #255: build error of ros_kortex in docker

patrickwasp, 2022-09-23:
## Description

ros_kortex 2.3.0 fails to build

## Version

ROS distribution : noetic

Branch and commit you are using : 2.3.0

## Steps to reproduce

1. run `docker build --file ros_kortex.dockerfile .`

## Code example (if necessary)

#### `ros_kortex.dockerfile`
```dockerfile
FROM osrf/ros:noetic-desktop-full

SHELL ["/bin/bash", "-c"]
ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get upgrade --yes

RUN apt-get update && apt-get --yes install \
    wget \
    python3 \
    python3-pip \
    python3-catkin-tools

RUN sudo python3 -m pip install conan==1.52.0

RUN mkdir -p /catkin_ws/src
WORKDIR /catkin_ws

RUN conan config set general.revisions_enabled=1 \
    && conan profile new default --detect > /dev/null \
    && conan profile update settings.compiler.libcxx=libstdc++11 default

RUN wget -c https://github.com/Kinovarobotics/ros_kortex/archive/refs/tags/v2.3.0.tar.gz -O - | tar -xz -C /catkin_ws/src
RUN mv /catkin_ws/src/ros_kortex-2.3.0 /catkin_ws/src/ros_kortex

RUN rosdep install --from-paths src --ignore-src -y

RUN catkin config --extend /opt/ros/$ROS_DISTRO \
    && catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release \
    && catkin build
```

## Expected behavior

Successful compile of ros_kortex package.

## Any other information

Buid output: 
```
Step 13/13 : RUN catkin config --extend /opt/ros/$ROS_DISTRO     && catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release     && catkin build
 ---> Running in 096b02f0110e
-----------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /catkin_ws
-----------------------------------------------
Build Space:       [missing] /catkin_ws/build
Devel Space:       [missing] /catkin_ws/devel
Install Space:      [unused] /catkin_ws/install
Log Space:         [missing] /catkin_ws/logs
Source Space:       [exists] /catkin_ws/src
[... 326 lines cut ...]
   77 | std::optional< MassMatrix3<T> > Ellipsoid<T>::MassMatrix() const
      |      ^~~~~~~~

...

/usr/include/sdformat-9.8/sdf/Param.hh: In member function ‘std::string sdf::v9::ParamPrivate::TypeToString() const [with T = double]’:
/usr/include/sdformat-9.8/sdf/Param.hh:320:3: warning: control reaches end of non-void function [-Wreturn-type]
make[2]: *** [CMakeFiles/roboticsgroup_gazebo_mimic_joint_plugin.dir/build.make:63: CMakeFiles/roboticsgroup_gazebo_mimic_joint_plugin.dir/src/mimic_joint_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:220: CMakeFiles/roboticsgroup_gazebo_mimic_joint_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /catkin_ws/build/roboticsgroup_gazebo_plugins; catkin build --get-env roboticsgroup_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
Failed <<< roboticsgroup_gazebo_plugins                         [ 10.1 seconds ]
[build 02:26.1 s] [10/14 complete] [1/16 jobs] [0 queued] [2 failed] [korte... 
...............................................................................
Failed << roboticsgroup_gazebo_plugins:make                     [ Exited with code 2 ]
Finished <<< kortex_driver                                        [ 2 minutes and 24.6 seconds ]
[build] Summary: 9 of 14 packages succeeded.                                   
[build] Ignored: None.                                                         
[build] Warnings: 2 packages succeeded with warnings.                          
[build] Abandoned: 3 packages were abandoned.                                  
[build] Failed: 2 packages failed.                                             
[build] Runtime: 2 minutes and 26.2 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.
The command '/bin/bash -c catkin config --extend /opt/ros/$ROS_DISTRO     && catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release     && catkin build' returned a non-zero code: 1
```

felixmaisonneuve, 2022-09-28:
Hi @patrickwasp,

I will link the issue #166 that seems to be the same errors as you. 
Basically, you are using a commit from before we added support for ROS Noetic
I will suggest changing 
```
https://github.com/Kinovarobotics/ros_kortex/archive/refs/tags/v2.3.0.tar.gz
```
to 
```
https://github.com/Kinovarobotics/ros_kortex/archive/refs/heads/noetic-devel.zip
```
in your dockerfile to download the latest commit from our noetic-devel

It sould fix your issue.

Regards,
Felix

felixmaisonneuve, 2022-10-28:
Hi @patrickwasp,

I will assume this fixed the issue and close it.
If it is not the case, feel free to re-open the issue or comment and I will help you

Regards,
Felix

closed 2022-10-28 by felixmaisonneuve