Kinovarobotics/ros_kortex #263: Is there an api to tell if the arm and its gripper is moving at any given time?

cjiang2, 2022-10-28:
Hello, I'm using the example script (https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_examples/src/full_arm/example_full_arm_movement.py#L213) in python3 to communicate with a real-world Kinova Gen3 and its gripper. However, when sending joint angle or a gripper value, there seems to be a slight delay (it's small but this delay before the actual movements is small enough to make a difference for my application) before the arm actually carries out the command. I'm wondering if there's something from the api to tell if the arm and its gripper is moving or executing a command at any given time?

I think the action notification (https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_examples/src/full_arm/example_full_arm_movement.py#L96) in some way, takes care of it. But it does not seem to account for gripper commands (https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_examples/src/full_arm/example_full_arm_movement.py#L276) though.

Thank you!

felixmaisonneuve, 2022-10-28:
Hi @cjiang2,

You are correct, the action notification does not account for gripper commands. 
There is no notification that indicates when the gripper command is cmpleted, so each gripper command has a `sleep` and that is probably where the delay you are seeing comes from (l.295) : 
https://github.com/Kinovarobotics/ros_kortex/blob/a46606e94f8437402cad91a7d7064f1c7f58ea91/kortex_examples/src/full_arm/example_full_arm_movement.py#L276-L296

You could implement a function that monitors the gripper's feedback end exit when the position is reached

Best,
Felix

cjiang2, 2022-10-28:
> Hi @cjiang2,
>
> You are correct, the action notification does not account for gripper commands. There is no notification that indicates when the gripper command is cmpleted, so each gripper command [...]

Hello, thank you for the reply. I will try something by monitoring the arm's gripper pose in this case. I will close the issue.

closed 2022-10-28 by cjiang2