Kinovarobotics/ros_kortex #265: How to change the velocity in cartesian space with REACH_POSE action ?

lexidliny, 2022-11-17:
Hello, I need change the velocity of the end effector when facing different situations in my project. I tried to modify the expample **example_cartesian_poses_with_notifications.cpp** to arrive at a fix pose with different velocitis. However, there was no change in the orientatioin speed, and the translation speed changed a little. 
```
  my_cartesian_speed.translation = 0.3f;
  my_cartesian_speed.orientation = 40.0f;
```
Is it because the soft limitation of the motor or the orientatioin and translation speed ? Now, it seems too slow to finish my work. How can I move the end effector faster in cartesian space like the above example ?

lexidliny, 2022-11-18:
Sorry, I will describe my problem in more detail. 
I want the robot to move from A to B, for example: 
```
  success &= example_home_the_robot(n, robot_name);
  success &= exampxample_cartesian_action(n, robot_name);
```
so that the robot first go home and than move to a designed pose. I want to change the velocity in the travel between the home and the designed pose like this: 
##velocity 1##
```
kortex_driver::ConstrainedPose my_constrained_pose;
  kortex_driver::CartesianSpeed my_cartesian_speed;

  my_cartesian_speed.translation = 0.3f;
  my_cartesian_speed.orientation = 40.0f;

  my_constrained_pose.constraint.oneof_type.speed.push_back(my_cartesian_speed);
```
##velocity 2##
```
  kortex_driver::ConstrainedPose my_constrained_pose;
  kortex_driver::CartesianSpeed my_cartesian_speed;

  my_cartesian_speed.translation = 0.f;
  my_cartesian_speed.orientation = 15.0f;

  my_constrained_pose.constraint.oneof_type.speed.push_back(my_cartesian_speed);
```
The designed pose is here:
```
my_constrained_pose.target_pose.x = 0.378;
  my_constrained_pose.target_pose.y = 0.005;
  my_constrained_pose.target_pose.z = 0.501;
  my_constrained_pose.target_pose.theta_x = 180.;
  my_constrained_pose.target_pose.theta_y = 0.;
  my_constrained_pose.target_pose.theta_z = 90.;
```
However, there is no different between the two action. The speed didn't change and it seemed to be slowly. How can I increase the speed? Thank you!

felixmaisonneuve, 2022-11-21:
I did some testing using something similar via the API (not ROS) and everything worked fine.
Your code looks good, so I am not sure why it is not working.  
Can you make sure you do not have software limits already set that could be slower than your constraints?

In the WebApp of the arm, go to "Speed Limits"
[image]

than "Advanced" and select "Cartesian Trajectory"
[image]

Make sure the linear limit is bigger than your constraints. (You could set them to max velocity limits).

If it doesn't work, could you monitor (roughly) the arm's tool twist linear velocity and check if it is lower than your constraint limits

lexidliny, 2022-12-02:
Thank you @felixmaisonneuve ! I have tried your method to set max the Speed Limits in the WebApp the same as your example. I tested two kinds of motion respectively which are translation and orientation. It is shown that the translation speed can meet my needs. However, the orentation speed can not arrive at the speed that I set in the `my_cartesian_speed.orientation`. Here I ran a pure rotate motion and  monitored the twist velocity:
**The first cartesian pose:**
```
kortex_driver::ConstrainedPose my_constrained_pose;
  kortex_driver::CartesianSpeed my_cartesian_speed;

  my_cartesian_speed.translation = 0.3f;
  my_cartesian_speed.orientation = 70.0f;

  my_constrained_pose.constraint.oneof_type.speed.push_back(my_cartesian_speed);

  my_constrained_pose.target_pose.x = 0.450;
  my_constrained_pose.target_pose.y = 0.018;
  my_constrained_pose.target_pose.z = 0.380;
  my_constrained_pose.target_pose.theta_x = 180.;
  my_constrained_pose.target_pose.theta_y = 0.;
  my_constrained_pose.target_pose.theta_z = 90.;
```

**The second cartesian pose:**
```
kortex_driver::ConstrainedPose my_constrained_pose;
  kortex_driver::CartesianSpeed my_cartesian_speed;

  my_cartesian_speed.translation = 0.3f;
  my_cartesian_speed.orientation = 70.0f;

  my_constrained_pose.constraint.oneof_type.speed.push_back(my_cartesian_speed);

  my_constrained_pose.target_pose.x = 0.450;
  my_constrained_pose.target_pose.y = 0.018;
  my_constrained_pose.target_pose.z = 0.380;
  my_constrained_pose.target_pose.theta_x = 180.;
  my_constrained_pose.target_pose.theta_y = 0.;
  my_constrained_pose.target_pose.theta_z = 0.;
```
The speed limits are set as here:
[image]

The monitor of the arm is here:
[image]
The orientation speed is far from what I set in the code. What's more, I found when a cartesian trajectiry combines with translation and orientation, the speed of translation speed is also effect by the orientation speed. It meas that the translation and orientation should begin at the same time and stop at the same time. When the orientation speed in a cartesian trajectory is slow, then translation speed also become slow. 

So, there is another problem now, how can I increase the orientation speed? It seems the max speed of orientation is not the same as the setting in the WebApp.

felixmaisonneuve, 2022-12-07:
You are using the deprecated `PlayCartesianTrajectory` API call, I would suggest to use Waypoints instead. I tested it and orientation speed limitations are working fine. 

Also, translation and orientation that begin and stop at the same time is expected. This is how it works. Angular movement have a similar behaviour as well (all joints start moving and stop a the same time).

felixmaisonneuve, 2023-02-22:
Hi @lexidliny 

can we close this issue?
Did you manage to make your limits work?

I will close this issue in a couple weeks if I do not get an answer from you

Regards,
Felix