Kinovarobotics/ros_kortex #266: catkin_make fails in ubuntu 18.04, ros melodic branch of kortex

apalapramanik, 2022-11-17:
[image]

felixmaisonneuve, 2022-11-17:
Hi @apalapramanik,
 
Can you give the output of `conan remote list` and `conan --version`

Best,
Felix

apalapramanik, 2022-11-17:
[image]

I restarted the computer and build my package again so catkin_make works fine now. Not sure what caused the problem. Attached are the command outputs asked for.

felixmaisonneuve, 2022-11-18:
Probably has to do with computer having to reboot after installing or changing some configuration in conan. 

If it works, I will close the issue

Best regards,
Felix

closed 2022-11-18 by felixmaisonneuve