Kinovarobotics/ros_kortex #267: Errors with real arm using Moveit

dufrenekm, 2022-11-21:
I have a 7 DOF Gen 3. I'm trying to use MoveIt for path planning, based on the provided MoveIt example. Often when I try to set a joint position or even `reach_named_position("home")` I get errors similar to the following: 
`[ INFO] [1669010873.351546066]: Solution found in 0.014242 seconds
[ INFO] [1669010873.367218342]: SimpleSetup: Path simplification took 0.015622 seconds and changed from 4 to 2 states
[ INFO] [1669010873.380658996]: New goal received.
[ INFO] [1669010873.380715707]: Joint Trajectory Goal is accepted.
[ERROR] [1669010873.413496807]: Joint Trajectory failed validation in the arm.
[ERROR] [1669010873.413573556]: Error 1 : Invalid Speed - Provided Waypoints Lead to go Over Maximum Speed on Actuator (5) and Segment (6).`

ROS distribution : Melodic, running in a docker container Ubuntu 18.04

Branch : melodic-devel, cloned 3 days ago.

This error was produced by modifying [example_move_it_trajectories.py](https://github.com/Kinovarobotics/ros_kortex/blob/melodic-devel/kortex_examples/src/move_it/example_move_it_trajectories.py). I added a second `reach_joint_angles` operation after the first. If I set it to a small change (.01-.03 radians), it works without issue. Any larger and these speed errors appear.

Here is the function I added. It is an exact copy of `reach_joint_angles` except joint 5 is set to rotate to .1 instead of pi/2.

`def reach_joint_angles2(self, tolerance):
    arm_group = self.arm_group
    success = True

    # Get the current joint positions
    joint_positions = arm_group.get_current_joint_values()
    rospy.loginfo("Printing current joint positions2 before movement :")
    for p in joint_positions: rospy.loginfo(p)

    # Set the goal joint tolerance
    self.arm_group.set_goal_joint_tolerance(tolerance)

    # Set the joint target configuration
    if self.degrees_of_freedom == 7:
      joint_positions[0] += 0
      joint_positions[1] += 0
      joint_positions[2] += 0
      joint_positions[3] += 0
      joint_positions[4] += 0
      joint_positions[5] = 0.1
      joint_positions[6] += 0
    elif self.degrees_of_freedom == 6:
      joint_positions[0] = 0
      joint_positions[1] = 0
      joint_positions[2] = pi/2
      joint_positions[3] = pi/4
      joint_positions[4] = 0
      joint_positions[5] = pi/2
    arm_group.set_joint_value_target(joint_positions)
    
    # Plan and execute in one command
    success &= arm_group.go(wait=True)

    # Show joint positions after movement
    new_joint_positions = arm_group.get_current_joint_values()
    rospy.loginfo("Printing current joint positions after movement :")
    for p in new_joint_positions: rospy.loginfo(p)
    return success`

felixmaisonneuve, 2022-11-21:
Hi @dufrenekm,

This seems to be a recurent error. I am pretty sure this has to do with the MoveIt Time Parametrisation. You could a try different Time Parametrisation algorithm, but others have tried and I am not sure they managed to make it work. 

The only workaround (without slowing the arm) seems to be to generate another trajectory.

I will link these other issues that cover the same topic : 
#206 
#233 
#258 

Best,
Felix

felixmaisonneuve, 2022-11-21:
I will close this issue and mark it as duplicate to #233.

If you have any other questions/comments, I will be happy to answer them and, if necessary, you can always re-open the issue

closed 2022-11-21 by felixmaisonneuve

felixmaisonneuve, 2022-11-24:
Hi @dufrenekm,

I will copy the message I wrote for the other issue.

Did you, at any point try to set the arm's velocity and acceleration software limits to the maximum value?
If you have software limits set, the error message would be the same, but I am not sure MoveIt handles software limits, so it might try to reach highier velocties/accelerations and causing errors.

To chnage your software velocity/acceleration limits, you can go in the WebApp, "Configurations"->"Speed Limits". You go to "Advanced" and select the "Cartesian Waypoint Trajectory" (but you might as well set the limits for every trajectory mode just to be safe). Then you click "Set Max" for every limit
[image]

zhangzhiyi22, 2024-06-05:
But the apply button is gray. What if I can't click it

martinleroux, 2024-06-05:
Hi @zhangzhiyi22 ,

Sounds like you may be trying to apply limits that are too large. If the issue persists, please contact us at support@kinova.ca

zhangzhiyi22, 2024-06-05:
so  if i still use the gen3_lite,how can i deal with the pronlem.

张志毅

南方科技大学/本科/本科2021级

广东省深圳市南山区学苑大道1088号