Kinovarobotics/ros_kortex #271: The problem when run the kortex and vision program on Gen3

YW0330, 2022-12-24:
## Description

One paragraph to explain the issue

## Version

ROS distribution :  ROS Noetic 
Kortex API: 2.5.0

## Steps to reproduce

1. roscore
2. [roslaunch kinova_vision kinova_vision_rgbd.launch](https://github.com/Kinovarobotics/ros_kortex_vision/blob/master/launch/kinova_vision_rgbd.launch)
3. rosrun ros_package my_program

## Code example (if necessary)
My code is modified from [01-torque_control_cyclic.cpp](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp)

## Expected behavior
When I following the steps described above. The terminal will show the warning message `runtime error: timeout detected: BaseCyclicClient::Refresh` first. After that, the error messages show as following:
```
Kortex exception: Device error, Error sub type=WRONG_SERVOING_MODE => <srv: 3, fct: 1, msgType: 3>
description: Wrong servoing mode, must be low level servoing mode
Error sub-code: WRONG_SERVOING_MODE
```
Then, my robot stops to receive my command.

## Any other information
I know the point cloud message in ros is very big data, so it will occupy the network.
I change the vision module resolution for the depth one is 420x240 and the color one is 320x240, but it doesn't improve this issue.

Thanks for helping.

felixmaisonneuve, 2023-01-02:
Hi @YW0330,

You need the ros_kortex driver to run to be able to use the ros_kortex_vision module
your are missing something like `roslaunch kortex_driver kortex_driver.launch ...`
Otherwise, your roscore is not connected to the arm API and every call you make will fail

Best,
Felix

YW0330, 2023-01-02:
Hi @felixmaisonneuve thank you for the response.

I found the problem, it cause from the Network. When I use Wi-Fi to connect Kinova Gen3, it will get the above issue. However, I change the connected method to Cable network, it can be useful. Have any idea to reduce the transmit data about depth information? Thanks a lot.

felixmaisonneuve, 2023-01-02:
I do not think you can reduce the transmit data other than using the smallest resolution setting. 
You can maybe use only color or only depth (depending on what your are doing) instead of both. 

Closing this issue since you found the issue

Regards,
Felix

closed 2023-01-02 by felixmaisonneuve