Kinovarobotics/ros_kortex #272: Angular Waypoint minimum duration

jabbershort, 2023-01-13:
## Summary

The Angular Waypoint command requires a duration. Any invalid (i.e. too short) duration will fail. Is it possible to set the duration to be 'minimum achievable' or similar?

## Use case

To allow for 'fastest' waypoint traversing when using angular waypoints. Currently Cartesian waypoints can achieve this by not setting max velocities.

## Alternative

Alternatively, when generating the message, the current position -> goal position rotations could be calculated and adjusted by achievable joint speeds to calculate duration, however this is relatively laborious.

felixmaisonneuve, 2023-01-13:
Hi @jabbershort,

Unfortunately, there is no way to specify a value for the arm to move in the fastest way (for waypoints). 
The workaround you provide could work to a certain extend, but I am not sure it will work on a reliable way. It is also very laborious like you said. 

Another workaround is to use the `ValidateWaypointList` call until it works : 
https://github.com/Kinovarobotics/ros_kortex/blob/6f9a44568eccb40048494a8b86b0dee7aec8c3fb/kortex_examples/src/full_arm/example_full_arm_movement.py#L246-L256

This is not so laborious in terms of code, but it adds a lot of computation time before moving the arm. 

You could use `ExecuteAction` or `Sequence` with `reach_joint_angles` actions to move in the fastest way, but the arm would have to stop at every point. You lose the advantage of `blending_radius` from waypoints

Best Regards,
Felix

jabbershort, 2023-01-15:
Hi @felixmaisonneuve,

Thanks for the reply, it's as I expected. As I'm actually coming from a different software stack, I will probably write something to estimate the time there and then give a bit of tolerance time.

Thanks!

closed 2023-01-16 by felixmaisonneuve