Kinovarobotics/ros_kortex #277: I first go from A to C, with tool(default){0,0,0.2,0,0,0}, then set tool {0,0.3,0.2,0,0,0} then go from C to D, with with tool(default){0,0.3,0.2,0,0,0},error!!!!!!!!!!!!

wentianheshi, 2023-02-01:
error:
[image]
code:
/*
 * KINOVA (R) KORTEX (TM)
 *
 * Copyright (c) 2019 Kinova inc. All rights reserved.
 *
 * This software may be modified and distributed
 * under the terms of the BSD 3-Clause license.
 *
 * Refer to the LICENSE file for details.
 *
 */

#include <thread>
#include <atomic>

#include "ros/ros.h"

#include <kortex_driver/Base_ClearFaults.h>
#include <kortex_driver/CartesianSpeed.h>
#include <kortex_driver/BaseCyclic_Feedback.h>
#include <kortex_driver/ReadAction.h>
#include <kortex_driver/ExecuteAction.h>

#include <kortex_driver/SetCartesianReferenceFrame.h>
#include <kortex_driver/CartesianReferenceFrame.h>

#include <kortex_driver/GetPayloadInformation.h>
#include <kortex_driver/SetToolConfiguration.h>

#include <kortex_driver/SendGripperCommand.h>
#include <kortex_driver/GripperMode.h>
#include <kortex_driver/ActionNotification.h>
#include <kortex_driver/ActionEvent.h>
#include <kortex_driver/OnNotificationActionTopic.h>
#include <kortex_driver/ExecuteWaypointTrajectory.h>
#include <kortex_driver/ValidateWaypointList.h>
#include <kortex_driver/GetProductConfiguration.h>
#include <kortex_driver/ModelId.h>

#include <math.h>

#include <kortex_driver/SetToolConfiguration.h>

#include <angles/angles.h>

#include "actionlib/client/simple_action_client.h"

#include "kortex_driver/BaseCyclic_Feedback.h"
#include "kortex_driver/ReadAction.h"
#include "kortex_driver/ExecuteAction.h"
#include "kortex_driver/ActionNotification.h"
#include "kortex_driver/ActionEvent.h"
#include "kortex_driver/OnNotificationActionTopic.h"
#include <kortex_driver/CartesianReferenceFrame.h>
#include "kortex_driver/FollowCartesianTrajectoryAction.h"
#include "kortex_driver/FollowCartesianTrajectoryActionFeedback.h"
#include "kortex_driver/FollowCartesianTrajectoryActionGoal.h"
#include "kortex_driver/FollowCartesianTrajectoryActionResult.h"
#include <kortex_driver/GetProductConfiguration.h>
#include <kortex_driver/ModelId.h>

#define HOME_ACTION_IDENTIFIER 2

std::atomic<int> last_action_notification_event{0};

void notification_callback(const kortex_driver::ActionNotification& notif)
{
[... 629 lines cut ...]

  //*******************************************************************************
  // Move the robot using Cartesian waypoint.
  // success &= example_cartesian_waypoint_action(n, robot_name);
  //*******************************************************************************
  //again
  // success &= example_send_cartesian_pose(n, robot_name,0.05,0,0.14,0,100,0);
  
  // success &= example_send_joint_angles(n, robot_name, degrees_of_freedom);

  // success &= example_send_cartesian_pose(n, robot_name,0,0,-0.1,0,0,0);

  // success &= example_send_gripper_command(n, robot_name, 0);

  // success &= example_send_joint_angles(n, robot_name, degrees_of_freedom);
  //*******************************************************************************
  // Move the robot to the Home position one last time.
  // success &= example_home_the_robot(n, robot_name);
  //*******************************************************************************

  // Report success for testing purposes
  ros::param::set("/kortex_examples_test_results/full_arm_movement_cpp", success);
  
  return success ? 0 : 1;
}

felixmaisonneuve, 2023-02-01:
Hi @wentianheshi,

I won't debug a 700 line file you wrote. Maybe you could guide me to the specific section you are having trouble with.

Based on your terminal output, your waypoint list is invalid. You can use the `ValidateWaypointList` function and print the errors in your waypoint list.

Best,
Felix

wentianheshi, 2023-02-02:
sorry  Felix 
I want say :
I set three waypoint A,B,C,and let robot move A to B to C is successed 
code:
    success &= example_send_cartesian_pose(n, robot_name, 0.1, 0, 0, 0, 0, 0);
    //success &= example_set_tool_transform(n, robot_name);
    success &= example_send_cartesian_pose(n, robot_name, 0, 0, 0, 0, 0, 0);
    success &= example_send_cartesian_pose(n, robot_name, -0.1, 0, 0, 0, 0, 0);
I set tool transform(0,0.1,0,0,0,0) successed 
code :
success &= example_set_tool_transform(n, robot_name);
but I set tool transform(0,0.1,0,0,0,0) in the process of moving A to B to C will error(robot can not move successed )
code:
     success &= example_send_cartesian_pose(n, robot_name, 0.1, 0, 0, 0, 0, 0);
    success &= example_set_tool_transform(n, robot_name);
    success &= example_send_cartesian_pose(n, robot_name, 0, 0, 0, 0, 0, 0);
    success &= example_send_cartesian_pose(n, robot_name, -0.1, 0, 0, 0, 0, 0);
error log :
Error Sending: Network is unreachablesendto() failed with error code : 101Error Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableError Sending: Network is unreachableE[ERROR] [1675303323.249223689]: Attempt to get goal status on an uninitialized ServerGoalHandle or one that has no ActionServer associated with it.
[ERROR] [1675303323.249256988]: Attempt to get goal status on an uninitialized ServerGoalHandle or one that has no ActionServer associated with it.
[ERROR] [1675303527.187535959]: Attempt to get goal status on an uninitialized ServerGoalHandle or one that has no ActionServer associated with it.
[ERROR] [1675303527.187593413]: Attempt to get goal status on an uninitialized ServerGoalHandle or one that has no ActionServer associated with it.

wentianheshi, 2023-02-02:
i also test code 
success &= example_send_cartesian_pose(n, robot_name, 0.1, 0, 0, 0, 0, 0);
success &= example_send_cartesian_pose(n, robot_name, 0, 0, 0, 0, 0, 0);
success &= example_send_cartesian_pose(n, robot_name, -0.1, 0, 0, 0, 0, 0);
success &= example_set_tool_transform(n, robot_name);
this process is successed

felixmaisonneuve, 2023-02-06:
Hi @wentianheshi,

So I have tried moving the arm between 3 poses like you did and setting your tool configuration between the second and third action.
Everything worked great for me. No ACTION_ABORT.

Looking at your log, I find it strange that you received an ACTION_ABORT for action 1001. You received an ACTION_END for action 1001 (Pose 1) just before. You then send Pose 12 to the arm, so I would expect a different action identifier. Are you sure about this piece of code?

Also, have you tried using other poses?

Best,
Felix

wentianheshi, 2023-02-07:
Hi @Felix
First，Thank you for your reply.
can you send me a copy of your testing program, I can compare the differences between the two, maybe this may help me the most.

felixmaisonneuve, 2023-02-07:
Yes, I just tested it and changing the tool configuration during an action will lead to an action abort. You have to wait until the action is finished, then change the tool configuration, then start your new action. 

You cannot do both at the same time

closed 2023-02-14 by wentianheshi