Kinovarobotics/ros_kortex #279: No conan binaries available for kortex_api_cpp/2.3.0-r.34@kortex/stable

gergondet, 2023-02-09:
## Description

I am trying to install this stack on Ubuntu 20.04 with ROS noetic, using the conan approach. It fails to find matching kortex_api_cpp binaries. I have done the required conan setup.

I get the following message:
```
-- +++ processing catkin package: 'kortex_driver'                                                                                                                                                                                                                                                        [23/491]
-- ==> add_subdirectory(ros_kortex/kortex_driver)                                                                                                                                                                                                                                                                
CONAN_TARGET_PLATFORM is x86                                                                                                                                                                                                                                                                                     
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy                                                                                
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system 
-- Generating .msg files for action kortex_driver/FollowCartesianTrajectory /home/gergondet/devel/src/catkin_data_ws/src/ros_kortex/kortex_driver/action/non_generated/FollowCartesianTrajectory.action
-- kortex_driver: 432 messages, 259 services                                                                                                                                                                                                                                                                     
-- Conan: checking conan executable in path                                                                                                             
-- Conan: Found program /home/gergondet/.local/bin/conan
-- Conan: Version found Conan version 1.57.0                                                                                                            
                                                                            
-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake                 
-- Conan: Settings= -s;build_type=RelWithDebInfo;-s;compiler=gcc;-s;compiler.version=9;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /home/gergondet/devel/src/catkin_data_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=RelWithDebInfo -s compiler=gcc -s compiler.version=9 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:comp
iler.libcxx=libstdc++11 -g=cmake --update                                   
Configuration:                                                              
[settings]                                                                  
arch=x86_64                                                                 
arch_build=x86_64                                                           
build_type=RelWithDebInfo                                                   
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=9                                                                                                                                                                                                                                                                                               
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]

conanfile.py: Installing package
Requirements
    kortex_api_cpp/2.3.0-r.34@kortex/stable from 'kinova_public' - Cache
Packages
    kortex_api_cpp/2.3.0-r.34@kortex/stable:72b117d1826364306c5396ab6067d2f7e3185825 - Missing

Installing (downloading, building) binaries...

ERROR: Missing binary: kortex_api_cpp/2.3.0-r.34@kortex/stable:72b117d1826364306c5396ab6067d2f7e3185825
kortex_api_cpp/2.3.0-r.34@kortex/stable: WARN: Can't find a 'kortex_api_cpp/2.3.0-r.34@kortex/stable' package for the specified settings, options and dependencies:
- Settings: arch=x86_64, build_type=RelWithDebInfo, compiler=gcc, compiler.libcxx=libstdc++11, compiler.version=5, os=Linux
- Options: shared=False
- Dependencies: 
- Requirements: 
- Package ID: 72b117d1826364306c5396ab6067d2f7e3185825

ERROR: Missing prebuilt package for 'kortex_api_cpp/2.3.0-r.34@kortex/stable'
Use 'conan search kortex_api_cpp/2.3.0-r.34@kortex/stable --table=table.html -r=remote' and open the table.html file to see available packages
Or try to build locally from sources with '--build=kortex_api_cpp'

More Info at 'https://docs.conan.io/en/latest/faq/troubleshooting.html#error-missing-prebuilt-package'

CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:72 (conan_cmake_run)
```

Indeed, if I execute: `conan search kortex_api_cpp/2.3.0-r.34@kortex/stable --table=out.html -r kinova_public` as suggested by conan, the table is empty.

[... 59 lines cut ...]
+                    SETTINGS kortex_api_cpp:arch=armv7 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=6.4 kortex_api_cpp:compiler.libcxx=libstdc++11 kortex_api_cpp:build_type=${CONAN_BUILD_TYPE}
                     ENV TARGET=imx6)
   elseif("${CONAN_TARGET_PLATFORM}" STREQUAL "jetson")
     conan_cmake_run(CONANFILE conanfile.py
                     UPDATE
                     BASIC_SETUP CMAKE_TARGETS
                     NO_OUTPUT_DIRS
-                    SETTINGS kortex_api_cpp:arch=armv7 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=7 kortex_api_cpp:compiler.libcxx=libstdc++11
+                    SETTINGS kortex_api_cpp:arch=armv7 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=7 kortex_api_cpp:compiler.libcxx=libstdc++11 kortex_api_cpp:build_type=${CONAN_BUILD_TYPE}
                     ENV TARGET=jetson)
   endif()
 endif()
diff --git a/kortex_driver/conanfile.py b/kortex_driver/conanfile.py
index e30328f..f5eeb31 100644
--- a/kortex_driver/conanfile.py
+++ b/kortex_driver/conanfile.py
@@ -6,4 +6,4 @@ import time
 class ROSKortexConan(ConanFile):  
 
     def requirements(self):
-        self.requires("kortex_api_cpp/2.3.0-r.34@kortex/stable")
+        self.requires("kortex_api_cpp/2.5.0-r.6@kortex/stable")
```

I would be happy to contribute a PR if that's ok with you?

felixmaisonneuve, 2023-02-13:
Hi @gergondet 

If I understand correctly, it won't find the good conan package because your `build_type` is not `Release` or `Debug`. If you want to keep `RelWithDebInfo` in your conan profile, can't you simply add `kortex_api_cpp:build_type=Release` to your settings?

I understand your modifications, but I don't think this is necessary

gergondet, 2023-02-14:
Hi @felixmaisonneuve 

Thanks for your reply. The problem is two fold:
- there's no binary available for kortex_api_cpp/2.3.0-r.34@kortex/stable only kortex_api_cpp/2.5.0-r.6@kortex/stable
- I think conan's build_type is decided based upon the `CMAKE_BUILD_TYPE` that is passed to catkin (at least that seems to be the case in the default configuration) but binaries are only available for Debug and Release types

> can't you simply add kortex_api_cpp:build_type=Release to your settings?

I could but:
- I would like it to work without extra steps if possible
- I don't want to remember to set that setting correctly if I decide to switch to a Debug build

Since CMake already knows which build type I want it seems reasonable to override the configuration there

felixmaisonneuve, 2023-02-16:
But binaries for for kortex_api_cpp/2.3.0-r.34@kortex/stable are available.
Here is the [Debug](https://artifactory.kinovaapps.com/ui/repos/tree/General/conan-public/kortex/kortex_api_cpp/2.3.0-r.34/stable/4a7af7ea3b5ba58e9864f879472e9d7059cd624f/package/9755f981ac44041fa0948f05e49abfa3d3346131/03dcc8509935528163d6c79e6f4ae128) conan package
Here is the [Release](https://artifactory.kinovaapps.com/ui/repos/tree/General/conan-public/kortex/kortex_api_cpp/2.3.0-r.34/stable/4a7af7ea3b5ba58e9864f879472e9d7059cd624f/package/c023db9fc677d4d0b3bd0c20f71385e4cf8a1220/31f1328bb975bbcb91651afbbd827a58) conan package

Looking at the error message you have, it says
```
WARN: Can't find a 'kortex_api_cpp/2.3.0-r.34@kortex/stable' package for the specified settings, options and dependencies:
```
and looking at th settings, you have `build_type=RelWithDebInfo`. 
```
- Settings: arch=x86_64, build_type=RelWithDebInfo, compiler=gcc, compiler.libcxx=libstdc++11, compiler.version=5, os=Linux
```

That's why I said your build_type is incorrect. We only have conan packages for `Release` and `Debug`.

You are right when you say it uses `CMAKE_BUILD_TYPE`, but unless you specify it with `-DCMAKE_BUILD_TYPE=Release/Debug`, I assume it will use what is set in your conan profile.

gergondet, 2023-02-16:
> But binaries for for kortex_api_cpp/2.3.0-r.34@kortex/stable are available.

My bad. Using the correct build type for the conan package lets conan correctly fetch the package (despite `conan search` reporting an empty page?)

> You are right when you say it uses CMAKE_BUILD_TYPE, but unless you specify it with -DCMAKE_BUILD_TYPE=Release/Debug, I assume it will use what is set in your conan profile.

It doesn't. My conan profile defaults to Release:
```
❯ conan profile show default
Configuration for profile default:

[settings]
os=Linux
os_build=Linux
arch=x86_64
arch_build=x86_64
compiler=gcc
compiler.version=9
compiler.libcxx=libstdc++
build_type=Release
[options]
[conf]
[build_requires]
[env]
```

When run via conan.cmake the `build_type` settings is automatically set to match the `CMAKE_BUILD_TYPE` I'm passing to catkin, hence, imho, it would make to either:
- pass `kortex_api_cpp:build_type` based on the current `CMAKE_BUILD_TYPE` so that the closest match is used
- document that only Release and Debug are supported

felixmaisonneuve, 2023-04-06:
I suppose we can close this issue now that the pull request is merged

closed 2023-04-06 by gergondet