Kinovarobotics/ros_kortex #281: Cannot install ros_kortex ERROR: Failed requirement 'kortex_api_cpp/2.3.0-r.34@kortex/stable' from 'conanfile.py'

lingxiaomeng, 2023-02-26:
-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=9;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /home/mlx/mr_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=9 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=9
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]

ERROR: Failed requirement 'kortex_api_cpp/2.3.0-r.34@kortex/stable' from 'conanfile.py'
ERROR: HTTPSConnectionPool(host='artifactory.kinovaapps.com', port=443): Max retries exceeded with url: /artifactory/api/conan/conan-public/v1/ping (Caused by ResponseError('too many 502 error responses')). [Remote: kinova_public]
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:72 (conan_cmake_run)

ch1hha0, 2023-02-26:
I had the same problem.

pradyum, 2023-02-26:
Hello, please check if the computer is connected to the internet. While building conan connects to https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public kinova public remote repository. The error is produced when it couldn't able to connect with the kinova public remote repository.

lingxiaomeng, 2023-02-27:
Thanks, there is no problem now. This may be caused by some unknown network problem.

closed 2023-02-27 by lingxiaomeng