Kinovarobotics/ros_kortex #282: Conan 2.0 not compatible with README instructions

leander-dsouza, 2023-02-27:
## Description

The [build](https://github.com/Kinovarobotics/ros_kortex#build) installation section fails to run due to the latest release of Conan 2.0.

## Version

ROS distribution: Noetic

Branch and commit you are using: noetic-devel, 6f9a44568eccb40048494a8b86b0dee7aec8c3fb

## Steps to reproduce
The execution of the following code block mentioned in the [README](https://github.com/Kinovarobotics/ros_kortex#build) causes errors.

```
conan config set general.revisions_enabled=1
conan profile new default --detect > /dev/null
conan profile update settings.compiler.libcxx=libstdc++11 default
```
## Expected behaviour

The profile creation of conan should be seamless with the latest version as well.

## Any other information

A quick fix would be specifying the latest compatible version in the installation of conan with this repository.
```
sudo python3 -m pip install conan==1.59.0
```

felixmaisonneuve, 2023-03-01:
Hi @leander-dsouza,

Commands changed between conan1.X and conan2.X, so I added the instructions to install conan==1.59.0 like you suggested. If users want/need conan2.X, they can always follow the [Conan migration guide](https://docs.conan.io/1/conan_v2.html).

I opened two identical Pull requests (one for the noetic branch and one for the melodic branch). You are free to comment if you feel something is incorrect.

This issue will be closed when the PRs will be merged

Regards,
Felix

leander-dsouza, 2023-03-02:
Hello @felixmaisonneuve,

The PRs suffice the Conan issue for now. In the future, probably the README will likely contain commands to ensure conan2.X compatibility.

PR #284 (2023-03-20): add note to indicate instructions will only work for conan versions 1.X
resolves #282 

Same as #283, but for the melodic branch
closed 2023-03-20 by felixmaisonneuve