Kinovarobotics/ros_kortex #286: rive gen3 using PreComputedJointTrajectory

ch1hha0, 2023-03-10:
Excuse me, does release 2.3.0 support using PreComputedJointTrajectory to drive gen3 7dof in ROS? I want to control the robotic arm in joint space, and I have written a program using PreComputedJointTrajectory to drive gen3. However, gen3 did not move, and there was no error message when the program was running. The following is part of the code:

`    def FillPreComputedJointTrajectory(self,point):
        precomputedjointWaypoint = kortex_driver.msg.PreComputedJointTrajectoryElement()

        precomputedjointWaypoint.joint_angles = point.positions
        precomputedjointWaypoint.joint_speeds = point.velocities
        precomputedjointWaypoint.joint_accelerations = point.accelerations
        precomputedjointWaypoint.time_from_start = point.time_from_start.secs + point.time_from_start.nsecs / 1000000000
        #precomputedjointWaypoint.time_from_start.sec = point.time_from_start.secs
        #precomputedjointWaypoint.time_from_start.usec = point.time_from_start.nsecs / 1000

        return precomputedjointWaypoint

    def example_send_pre_computed_joint_trajectory(self,plan):
        self.last_action_notif_type = None
        #req = kortex_driver.srv.PlayPreComputedJointTrajectoryRequest()
        req = kortex_driver.srv.ExecuteActionRequest()
        goal = kortex_driver.msg.PreComputedJointTrajectory()
        #goal.mode = 0
        for point in plan.joint_trajectory.points:
            #req.input.trajectory_elements.append(self.FillPreComputedJointTrajectory(point))
            goal.trajectory_elements.append(self.FillPreComputedJointTrajectory(point))

        req.input.oneof_action_parameters.play_pre_computed_trajectory.append(goal)
        
        #print(req.input.trajectory_elements)
        rospy.loginfo("Sending the PreComputedJointTrajectory...")
        try:
            #self.play_pre_computed_joint_trajectory(req)
            self.execute_action(req)
        except rospy.ServiceException:
            rospy.logerr("Failed to call PlayPreComputedJointTrajectory")
            return False
        else:
            return self.wait_for_action_end_or_abort()`

The `plan` is computed using MoveIt to generate a path. If this control method works well, I plan to explore other algorithms to generate paths in the future.

felixmaisonneuve, 2023-03-17:
Hi @ch1hha0,

You can refer to the old joint trajectory server that was using PlayPreComputedJointTrajectory to execute MoveIT trajectories
https://github.com/Kinovarobotics/ros_kortex/blob/0b422bb0770b2383eac4dec2d1566b3d65c3c31f/kortex_driver/src/non-generated/driver/pre_computed_joint_trajectory_action_server.cpp#L91-L131

Quickly looking at your code, I think you are missing a [mode](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/enums/Base/TrajectoryContinuityMode.md#) to your [PreComputedJointTrajectory](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/Base/PreComputedJointTrajectory.md#)
```py
goal = kortex_driver.msg.PreComputedJointTrajectory()
goal.mode = kortex_driver.msg.TrajectoryContinuityMode.TRAJECTORY_CONTINUITY_MODE_POSITION # not sure if this is the correct syntax
```

If you don't specify this field, it will defaults to 0, which is "TRAJECTORY_CONTINUITY_MODE_UNSPECIFIED"
It might explain why it is not working

Regards,
Felix

ch1hha0, 2023-03-23:
Hi @felixmaisonneuve ,
I tried your suggestion, goal.mode = `kortex_driver.msg.TrajectoryContinuityMode.TRAJECTORY_CONTINUITY_MODE_POSITION` , 
but the robotic arm still cannot move and there are no error messages. I wonder if there are other ways to execute joint trajectories continuously, such as FollowJointTrajectoryAction, but it seems to be disabled. I also tried Waypointlist, but this method did not meet my expectations.

felixmaisonneuve, 2023-04-03:
You can use `PlayJointTrajectory`. The fonction is deprecated and will maybe be removed in the future, but it is still working at the moment. You can also do a `reach_joint_angles` action. I think these are your only 2 options other than waypoint lists and PlayPreComputedJointTrajectory. 

I am not sure why your `PreComputedJointTrajectory` action does not work. What des your trajectory look like? You should have a trajectory element for every ms. The trajectory has to be less than 30s (30 000 trajectory elements) if I am not mistaken. Also, your first trajectory element should be the starting point of the arm.

ch1hha0, 2023-04-10:
The trajectory I am currently using for testing is shown below. In simple terms, I am trying to use the robotic arm to perform the task of writing.
```
[positions: [7.410547998354389e-06, 0.2601718368963022, 3.1400228969698714, -2.2701010041961975, -3.1287399238522084e-05, 0.9597233990196008, 1.5697763787858365]
velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
accelerations: [-0.02918398815095891, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: []
time_from_start: 
  secs: 0
  nsecs:         0, positions: [-0.001767933568192587, 0.260203081337934, 3.1403335499175133, -2.270120268050883, -0.0036417485157495314, 0.9595270069079064, 1.5732141880022623]
velocities: [0.00469403990733471, 0.03897158291989309, 0.0020930239964484904, -0.0016976423044843271, 0.005483072362160208, 0.04031626345688822, -0.0014617296767829874]
accelerations: [0.053545272804457145, 0.21279507965251795, 0.006580577235939805, -0.008988673865472242, 0.08665690674839718, 0.22372597056381188, -0.06193978823672886]
effort: []
time_from_start: 
  secs: 0
  nsecs: 348806003, positions: [0.003763702984397301, 0.28994904635449564, 3.141592653589793, -2.271396422519474, 0.0045029818031776985, 0.9905498603442838, 1.5683314920472553]
velocities: [0.022079406569185674, 0.1083744621356984, 0.0016477149346538855, -0.0022973701371706177, 0.02786756521510096, 0.11073440429430251, -0.015118131527508814]
accelerations: [0.05086052667876921, 0.20420930955888741, -0.011024545182339753, 0.0069764156285580395, 0.04382818142763983, 0.19763869858703123, -0.015648044113166917]
effort: []
time_from_start: 
  secs: 0
  nsecs: 730881702, positions: [0.010167647836953507, 0.31991701613054674, 3.141592653589793, -2.2716671322751147, 0.011929003172292932, 1.020815126411263, 1.5645650090020007]
velocities: [0.03304814957586543, 0.15737392226093003, 0.0, 0.001429539026842458, 0.03759316977550625, 0.15613904009160323, -0.01798568084139683]
accelerations: [0.03481079890704677, 0.19103589709137941, 0.0, 0.02775011146238188, 0.03282489943457213, 0.16402428439631309, -0.00546697933454536]
effort: []
time_from_start: 
  secs: 0
  nsecs: 946641073, positions: [0.016400281173262605, 0.3500148867430903, 3.141592653589793, -2.270963045269504, 0.01890665769353381, 1.0502544525089772, 1.5613961784537491]
velocities: [0.03895971177843781, 0.1911023463821344, 0.0, 0.007786108905702407, 0.042913304120413304, 0.18364874350856922, -0.018288783440643448]
accelerations: [0.03206926662653067, 0.1922090295879533, 0.0, 0.046286288227102165, 0.027036044840147603, 0.14675984152925942, 0.0028448418616559873]
effort: []
time_from_start: 
  secs: 1
  nsecs: 117795129, positions: [0.02246839992268003, 0.38018467888286916, 3.141592653589793, -2.269287757732766, 0.02549442785722757, 1.0788072815582974, 1.5587552604568629]
velocities: [0.04373903908671848, 0.2206001069611367, 0.0, 0.016007864060829987, 0.04683039051138613, 0.20505065125027022, -0.017490815792792862]
accelerations: [0.032538320241834395, 0.20743291686091547, 0.0, 0.06623399346117247, 0.025798959103924164, 0.14206498245745597, 0.008331403585530153]
effort: []
time_from_start: 
  secs: 1
  nsecs: 264000475, positions: [0.02837800170758121, 0.4103725781705975, 3.141592653589793, -2.2666452793073733, 0.03174189600446861, 1.1064192738540501, 1.5565805106618176]
velocities: [0.047504123628528946, 0.24584893278654663, 0.0, 0.025624630688364958, 0.049639984122069256, 0.22078631133010518, -0.015971377562222083]
accelerations: [0.024885567461499485, 0.17891552356929186, 0.0, 0.08241407983747955, 0.016874303869879117, 0.09721890073341165, 0.015404664625368]
effort: []
time_from_start: 
  secs: 1
  nsecs: 392542746, positions: [0.03413481844643665, 0.4405277731894516, 3.141592653589793, -2.2630419173308445, 0.03769163983181865, 1.1330422681776098, 1.5548166099281637]
velocities: [0.050198389306180084, 0.2661138103351389, 0.0, 0.03620298457548662, 0.051381810956827145, 0.23054734920987824, -0.013934445034906076]
accelerations: [0.020548010355091424, 0.16351699658224417, 0.0, 0.09727371144126035, 0.012431203755840975, 0.06677987521322942, 0.019235052347891763]
effort: []
time_from_start: 
  secs: 1
  nsecs: 509946735, positions: [0.039744329696457234, 0.4706029055141752, 3.141592653589793, -2.2584861633026145, 0.04338017488464592, 1.1586344839342804, 1.5534137886313983]
velocities: [0.05236168323746018, 0.283848428394917, 0.0, 0.04768465379331538, 0.052680235952870726, 0.23684971431662977, -0.01172523875266328]
accelerations: [0.01888087581698637, 0.16006817708274826, 0.0, 0.11282674485180523, 0.011227377505870351, 0.04760077438326669, 0.021154113117187373]
effort: []
time_from_start: 
  secs: 1
  nsecs: 619160817, positions: [0.045211775103939617, 0.5005544311928128, 3.141592653589793, -2.2529885441122075, 0.04883876119695438, 1.1831606214254364, 1.5523271012506643]
velocities: [0.054168541832349565, 0.29977467505635125, 0.0, 0.05997137483118589, 0.0537382720619437, 0.24051613179604658, -0.009486544945992319]
accelerations: [0.016201041777102528, 0.14954590808632795, 0.0, 0.12668843972972427, 0.009304891066915852, 0.02301544346949401, 0.022449925755247208]
effort: []
time_from_start: 
  secs: 1
  nsecs: 721622575, positions: [0.05054216621286735, 0.5303428894739095, 3.141592653589793, -2.2465614429186216, 0.05409409927745623, 1.2065918569204706, 1.5515158234367885]
velocities: [0.055850354193010446, 0.3150661962496869, 0.0, 0.07320638394926571, 0.054790584227746006, 0.24258156616443, -0.007309389085047071]
[... cut ...]
```
[ERROR] [1680496495.286302327]: Runtime exception detected while sending the trajectory
[ERROR] [1680496495.286365021]: timeout detected: BaseClient::ValidateWaypointList

[ERROR] [1680496500.331258774]: Runtime exception detected while sending the trajectory
[ERROR] [1680496500.331306895]: timeout detected: BaseClient::ExecuteWaypointTrajectory

[ERROR] [1680496500.331338189]: To transition to an aborted state, the goal must be in a preempting or active state, it is currently in state: 4
```

ch1hha0, 2023-04-11:
Additionally, I also set the api_rpc_timeout_ms to 30000 based on the suggestion in #278 , but the error still persists.

felixmaisonneuve, 2023-05-10:
Hi @ch1hha0,

Have you tried using `PreComputedJointTrajectory` with a smaller amount of points?
Does the PreComputedTrajectory you send seem correct? (your `#print(req.input.trajectory_elements)` line in you original message)

If you are using the matlab API, I would suggest to look at our [github repo](https://github.com/Kinovarobotics/matlab_kortex) rather than using the MatLab documentation. However, our MatLab API is still at Kortex2.2.0, so waypoints are not available on it, but the functions you mentionned so far should be. 

I am sure I fully understand what you are doing to get the error message. Are you using MatLab or FollowJointTrajectoryAction?

Best,
Felix

ch1hha0, 2023-05-23:
Sorry, I haven't been using this arm recently, so I haven't been able to respond. But what I can confirm is that I have tried using a `PreComputedJointTrajectory` with a few points before, but the arm still couldn't move. And, the first point of the trajectory I used is indeed the starting point of the arm. There were no error messages either. Additionally, could you provide an example code for a runnable `PreComputedJointTrajectory`? I believe this might help me resolve the issue faster. Thank you very much.

ch1hha0, 2023-06-21:
Thank you, Felix. I haven't been using this robotic arm recently, but I will continue to provide feedback if there are any new issues. I'm also looking forward to your adding an example for the `PreComputedJointTrajectory`.

ch1hha0, 2023-08-11:
Hello, I've been recently attempting to use `follow_joint_trajectory` to control a robotic arm. I've generated a trajectory that lasts for 4 seconds, with a time interval of 1 millisecond for each point. However, when I send this trajectory to `follow_joint_trajectory`, I encountered the following error:
```
[ERROR] [1691728694.274566359]: Runtime exception detected while sending the trajectory
[ERROR] [1691728694.274605140]: timeout detected: BaseClient::ValidateWaypointList
```
When I use MoveIt's `compute_cartesian_path` to generate a trajectory that lasts for 14 seconds and use `execute` to execute it, the following error occurs:
```
[ERROR] [1691729920.570040165]: Runtime exception detected while sending the trajectory
[ERROR] [1691729920.570139730]: timeout detected: BaseClient::ValidateWaypointList

[ INFO] [1691729920.570727564]: Controller 'gen3_joint_trajectory_controller' successfully finished
[ WARN] [1691729920.570940032]: Controller handle gen3_joint_trajectory_controller reports status ABORTED
[ INFO] [1691729920.571030258]: Completed trajectory execution with status ABORTED ...
[ INFO] [1691729920.571286200]: Execution completed: ABORTED
[ERROR] [1691729923.570473039]: Runtime exception detected while sending the trajectory
[ERROR] [1691729923.570524820]: timeout detected: BaseClient::ClearFaults

[ERROR] [1691729923.570565594]: To transition to an aborted state, the goal must be in a preempting or active state, it is currently in state: 4
```

I'd like to understand the correct way to use follow_joint_trajectory. The code I'm using is from [Trajectory control](https://github.com/Kinovarobotics/ros_kortex/issues/29#issuecomment-558583439), available at: https://github.com/akihikoy/ay_test/blob/master/ros/py_ros/gen3/follow_q_traj3.py
When using this code, I made necessary modifications to support Python 3, but I'm still encountering the aforementioned error.

ch1hha0, 2023-08-15:
Hello, I've tried using `PreComputedJointTrajectory` again, and I used the trajectory file ["trajectory.mat"](https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public%2Fkortex%2Fmatlab%2Fsimplified_API%2F2.2.1%2Fmatlab_simplified_api_2.2.1.zip) provided by `matlab_kortex`. Here's the code I used to read the .mat file and input it into PreComputedJointTrajectory:

```
def load_trajectory():
    mat = loadmat("trajectory.mat")
    pre_compt_traj = mat['precomputedTrajectory']
    times = pre_compt_traj[0:2,0].astype('float32')
    positions = np.rad2deg(pre_compt_traj[:,1:1 + 7]).astype('float32')
    velocity = np.rad2deg(pre_compt_traj[:, 8:8 + 7]).astype('float32')
    accelerations = np.rad2deg(pre_compt_traj[:, 15 : 15 + 7]).astype('float32')
    goal = kortex_driver.msg.PreComputedJointTrajectory()
    goal.mode = kortex_driver.msg.TrajectoryContinuityMode.TRAJECTORY_CONTINUITY_MODE_ACCELERATION
    for i in range(times.shape[0]):
        point = kortex_driver.msg.PreComputedJointTrajectoryElement()
        point.joint_angles = [x for x in positions[i]]
        point.joint_speeds = [x for x in velocity[i]]
        point.joint_accelerations = [x for x in accelerations[i]]
        point.time_from_start = times[i]
        goal.trajectory_elements.append(point)
    req = kortex_driver.srv.ExecuteActionRequest()
    req.input.oneof_action_parameters.play_pre_computed_trajectory.append(goal)
execute_action_full_name = '/' + self.robot_name + '/base/execute_action'
rospy.wait_for_service(execute_action_full_name)
execute_action = rospy.ServiceProxy(execute_action_full_name, kortex_driver.srv.ExecuteAction)
#req.input.handle.action_type = kortex_driver.msg.ActionType.PLAY_PRE_COMPUTED_TRAJECTORY
execute_action(req)
```
There are a total of 14601 trajectory points, which doesn't exceed the [hard limit](https://github.com/Kinovarobotics/matlab_kortex/blob/master/simplified_api/documentation/precomputed_joint_trajectories.md) of `PreComputedJointTrajectory`. Below is a portion of the trajectory points printed using `print(req)`:

```
input: 
  handle: 
    identifier: 0
    action_type: 0
    permission: 0
  name: ''
  application_data: ''
  oneof_action_parameters: 
    send_twist_command: []
    send_wrench_command: []
    send_joint_speeds: []
    reach_pose: []
    reach_joint_angles: []
    toggle_admittance_mode: []
    snapshot: []
    switch_control_mapping: []
    navigate_joints: []
    navigate_mappings: []
    change_twist: []
    change_joint_speeds: []
    change_wrench: []
    apply_emergency_stop: []
    clear_faults: []
    delay: []
    execute_action: []
    send_gripper_command: []
    send_gpio_command: []
    stop_action: []
    play_pre_computed_trajectory: 
      - 
        mode: 3
        trajectory_elements: 
          - 
            joint_angles: 
              - 0.0
              - 0.0
              - 0.0
              - 0.0
              - 0.0
              - 0.0
[... 247 lines cut ...]
              - 3.1483043e-05
              - 0.0010795941
              - 2.5152685e-05
              - 0.00017795275
            time_from_start: 0.009
    execute_sequence: []
    execute_waypoint_list: []
```

However, this still doesn't make the robotic arm move. The error message obtained using `GetTrajectoryErrorReport` is as follows:
```
output: 
  trajectory_error_elements: 
    - 
      error_type: 9
      error_identifier: 0
      error_value: 0.0
      min_value: 1.0
      max_value: 99999.9921875
      index: 0
      message: "Invalid Trajectory - Trajectory is empty"
      waypoint_index: 0
```

I've consulted [TrajectoryErrorType.md](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/enums/Base/TrajectoryErrorType.md), but I believe the trajectory should be valid since it's provided by [matlab_kortex](url). I've also tried changing the [goal.mode](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/enums/Base/TrajectoryContinuityMode.md) to other values and specifying `req.input.handle.action_type = kortex_driver.msg.ActionType.PLAY_PRE_COMPUTED_TRAJECTORY`. However, this still yields the same error message as mentioned above.

ch1hha0, 2023-08-28:
Hello, I have recently been using the Service method to use PreComputedJointTrajectory and have successfully moved the robot arm. However, I found that the description of the hard limits for [precomputed_joint_trajectories](https://github.com/Kinovarobotics/matlab_kortex/blob/master/simplified_api/documentation/precomputed_joint_trajectories.md) in [matlab_kortex](https://github.com/Kinovarobotics/matlab_kortex) (The acceleration limit of every joint must be respected for all the trajectory points. The trajectory will be rejected if any of the points contain an illegal acceleration. The hard acceleration limit for each joint is 1 degree / second ^ squared.) is incorrect. In the trajectory points provided in trajectory.mat, the maximum acceleration values for each joint are:
2.93391904269429	4.18121199820796	5.58093764832027	9.33660702926885	7.53745644870699	5.40291825949929	3.80256206887719
Therefore, I would like to know what the correct hard limits are when using PreComputedJointTrajectory.

shi00317, 2024-07-07:
Hi @ch1hha0,
I have the same problem. I am wondering are you successful using play_pre_compute_trajectory with python or Matlab? Here is my code: 
```python
def play_pre_compute(traj):
    goal = PreComputedJointTrajectory()
    goal.mode = kortex_driver.msg.TrajectoryContinuityMode.TRAJECTORY_CONTINUITY_MODE_POSITION
    for i in traj:
        joint_list = PreComputedJointTrajectoryElement()
        joint_list.joint_angles = i['positions']
        joint_list.joint_speeds = i['velocities']
        joint_list.joint_accelerations = i['accelerations']
        joint_list.time_from_start = i['time_from_start']['secs']+i['time_from_start']['nsecs']
        goal.trajectory_elements.append(joint_list)
    
    service = rospy.ServiceProxy('/my_gen3/base/play_pre_computed_joint_trajectory', PlayPreComputedJointTrajectory)
    resp1 = service(goal)
```

The pre compute trajectory is made my moveit:
```yaml
  positions: [-0.7116887886428893, 1.2937886912747407, -1.2954718090836703, -0.9481631123720398, 0.2918366823914488, -0.3021803667758034, 1.054209587767749]
  velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  accelerations: [9.237055564881302e-14, -2.418360650624507e-13, 2.3148150063434514e-13, -5.685035775471192e-14, -2.7200464103316335e-14, -3.147482274812319e-14, 2.6989521728637555e-13]
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
- 
  positions: [-0.7115911248425385, 1.293793204077657, -1.2953108883778615, -0.9481476267502637, 0.29164175465029113, -0.30203616995857485, 1.054325526879847]
  velocities: [0.00864197345499454, 0.0003993242416344238, 0.014239385145404182, 0.001370275699313958, -0.01724856455181653, 0.012759538972555842, 0.010259100358032958]
  accelerations: [0.5098013452173966, 0.023556660596731924, 0.8399999999999508, 0.08083436016872421, -1.01751543873402, 0.7527019339320946, 0.6051977815578]
  effort: []
  time_from_start: 
    secs: 0
    nsecs:  33903298
- 
  positions: [-0.6859682906275401, 1.2949771719404148, -1.2530921281833007, -0.944084857146164, 0.24050099238579642, -0.2642050479982637, 1.0847430268915985]
  velocities: [0.15685905637290967, 0.007248069443441325, 0.25845676671773077, 0.02487165162964606, -0.3130758933102061, 0.2315963192217543, 0.18621126410260352]
  accelerations: [0.46170088359478606, 0.02133405710688794, 0.7607448388631067, 0.0732075265489838, -0.9215114505719482, 0.6816834660013223, 0.5480965342994647]
  effort: []
  time_from_start: 
    secs: 0
    nsecs: 339032980
- 
  positions: [-0.6602477926121908, 1.2961656526060892, -1.2107124472829311, -0.9400066019202882, 0.18916530238014406, -0.22622972922072404, 1.115276466015448]
  velocities: [0.1987437246048692, 0.009183456478019892, 0.327470161140694, 0.03151290589304014, -0.39667374367306796, 0.293437409042446, 0.23593358934373923]
  accelerations: [0.13012036652521616, 0.006012540648811779, 0.21439941048915984, 0.02063195138981042, -0.259707988366871, 0.19211767965366094, 0.15446910428015811]
  effort: []
  time_from_start: 
    secs: 0
    nsecs: 480577976
... 

```
So I just wondering do you have any suggestion for my code.

Thanks.

ch1hha0, 2024-07-08:
Hi @shi00317 ,
Recently, I successfully used `play_pre_compute_trajectory` in Python, but the trajectory was derived from `matlab_kortex`. The trajectory I calculated myself still can not be executed. What I currently know is that the trajectory needs to be input at strict 1ms intervals. I suggest you refer to [01-actuator_low_level_velocity_control.cpp](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/200-Actuator_low_level_control/01-actuator_low_level_velocity_control.cpp) and use `low_level` to implement continuous trajectories.

martinleroux, 2024-07-08:
Hi all,

The robot not moving when trying to launch a PreComputed trajectory indicates that at least one point of the trajectory is not within the limits of the arm. 

As per our documentation in matlab_kortex, here are the validation that the robot does before launching the trajectory:
Once sent to the arm, a trajectory is validated and the arm will execute it if it meets all the conditions :

    The starting timestamp must be 0.0 seconds. Timestamps are all relative to the starting timestamp, which must be 0.0.

    Time increments must be 0.001 seconds between trajectory points. For now, the arm only accepts trajectories with waypoints every 1ms.

    The current position of the arm must match the first trajectory point. The trajectory planners communicating with the System object have to take into account the current position of the arm for the computation of the trajectories.

    The position limit of every joint must be respected for all the trajectory points. The trajectory will be rejected if any of the points contain an illegal position.

    The velocity limit of every joint must be respected for all the trajectory points. The trajectory will be rejected if any of the points contain an illegal velocity. The hard velocity limit for each joint is 50 degree / second..

    The acceleration limit of every joint must be respected for all the trajectory points. The trajectory will be rejected if any of the points contain an illegal acceleration. The hard acceleration limit for each joint is 1 degree / second ^ squared.  ****This is a mistake, the actual limit is 1 rad/s^2, as indicated in our user guide***

    The arrays must contain the same number of joints and number of trajectory points. The trajectory will be rejected if there is a mismatch in array sizes.

    Position continuity must be ensured for all trajectory points. For now, position continuity is the only supported trajectory continuity mode.

    The trajectory must not last more than 30 seconds. For now, it is the maximum supported size for a trajectory (30000 trajectory points).

______________________________

To find out which point(s) triggered the rejection of your trajectory, as indicated above, you should refer to the GetTrajectoryErrorReport service (see [here](https://github.com/Kinovarobotics/ros_kortex/blob/d57a287cdc85031ae351089b82c71cb8cace16be/kortex_driver/src/generated/robot/base_services.cpp#L205)).

It has been known for a while that the trajectories generated by MoveIt can often exceed the robot's acceleration limits. The issue has been solved in our [ROS2 implementation](https://github.com/Kinovarobotics/ros2_kortex) which uses our low-level interface instead of the PreComputedTrajectory endpoint. In these cases, MoveIt still exceeds the limits of the robot, but the robot provides a best effort until it ends up either lagging too far behind the commanded trajectory or, in most cases, it catches up.

As mentioned in @ch1hha0 's previous post, continuous trajectories can also be implemented directly using the low-level API outside of ROS. 

Hope this helps.

aalmrad, 2024-10-16:
Closing because of inactivity. Please reply to this thread to reopen.

Best,
Abed