Kinovarobotics/ros_kortex #290: Kinova Gen3 automatic robot boot on power-up

AlfaKeNTAvR, 2023-03-31:
## Summary

Is it possible to configure Kinova Gen3 to turn ON automatically when the power is supplied bypassing the power button?

## Use case

We are powering the arm from a mobile base, which has an internal breaker, which we control from a script. Once we apply power to the breaker it would be nice to automatically turn ON the arm without the need to physically press the power button.

felixmaisonneuve, 2023-03-31:
Hi @AlfaKeNTAvR,

Unfortunately, it is not possible to configure this on a Gen3 arm that does not have this feature.
We recently added a new Gen3 variant that comes with the automatic power ON variant, but it has a different hardware revision.

I can refer you to the "Powering on a robot conﬁgured with automatic startup" section in the user guide (revision 8) for more details. I can link you the [KortexAPI 2.5.0 release page](https://github.com/Kinovarobotics/kortex/releases/tag/v2.5.0) for more informations.

I will close this issue since this is not something you can do, but feel free to ask any question if something is unclear

Regards,
Felix

closed 2023-03-31 by felixmaisonneuve

AlfaKeNTAvR, 2025-07-01:
Hello, today we received a new Kinova Gen3 7DoF robot. Is it possible to power it up automatically without the quick release base? Where can we enable this functionality? It is not mentioned in the guide.

Best wishes,
Nikita

aalmrad, 2025-07-02:
@AlfaKeNTAvR, if the robot you have received includes the custom "Auto Power On" feature (which is normally specified upon ordering), then it will be automatically enabled and the robot is supposed to be turned on as soon as you plug its power supply cable to an adequate power source.

AlfaKeNTAvR, 2025-07-02:
@aalmrad it doesn't seem to have this option... Is there a way to enable it somehow post-purchase?