Kinovarobotics/ros_kortex #293: Joint limit mismatch in URDF vs Hardware

leander-dsouza, 2023-04-03:
## Description

* The software shipped with the hardware of the robot has different joint limits set in comparison to the limits set in the URDF.

* These are the limits set in the software shipped with the Gen3 arm hardware: [(pg 98)](https://www.kinovarobotics.com/uploads/User-Guide-Gen3-R07.pdf)

   [image]

* These are the limits set for actuators in the [urdf](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_description/arms/gen3/7dof/urdf/gen3_macro.xacro):

    * [Actuator 2](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_description/arms/gen3/7dof/urdf/gen3_macro.xacro#L99) (`+-138.08` deg):
    
          <limit lower="-2.41" upper="2.41" effort="39" velocity="1.3963" />

    * [Actuator 4](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_description/arms/gen3/7dof/urdf/gen3_macro.xacro#L169) (`+-152.40` deg):
    
          <limit lower="-2.66" upper="2.66" effort="39" velocity="1.3963" />

    * [Actuator 6](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_description/arms/gen3/7dof/urdf/gen3_macro.xacro#L239) (`+-127.76` deg):
    
          <limit lower="-2.23" upper="2.23" effort="9" velocity="1.2218" />

* Due to this mismatch, when MoveIt! plans paths it considers the urdf's limits instead of the hardware. As a result, occasionally, the following error message gets printed onto the console, when actuator 6 exceeds the hardware limits.

      [ERROR] : Joint Trajectory failed validation in the arm.
      [ERROR] : Error 1 : Invalid Joint Position - Provided joint position for actuator (5) exceeds the joint limits
      [ERROR] : Error 2 : Invalid Joint Position - Provided joint position for actuator (5) exceeds the joint limits
      [ERROR] : Error 3 : Invalid Joint Position - Provided joint position for actuator (5) exceeds the joint limits

## Version

ROS distribution: noetic

Branch and commit you are using: `noetic-devel`, adb3315aa90da73875762257a39ec4f60589f634

## Steps to reproduce

1. Run the hardware launch file for the Gen3 Kinova arm:

       roslaunch kortex_driver kortex_driver.launch

2. Add the MotionPlanning plugin and select the Joints tab. 

   [image]

3. Drag the slider of joint_6 to 128 degrees:

   [image]

4. Finally use the Planning Tab and `Plan and Execute`.

## Expected behavior

The joint limits of the urdf and the software shipped with the robot is supposed to be matched.

## Any other information

* A quick fix would be to correct the urdf values as follows:

  * Actuator 2 - (`+-128.9` deg):
  
        <limit lower="-2.24" upper="2.24" effort="39" velocity="1.3963" />
  
  * Actuator 4 - (`+-147.8` deg):
  
        <limit lower="-2.57" upper="2.57" effort="39" velocity="1.3963" />
  
  * Actuator 6 - (`+-120.3` deg):
  
        <limit lower="-2.09" upper="2.09" effort="9" velocity="1.2218" />

* This may solve the following issue as well - #245.

felixmaisonneuve, 2023-04-03:
Hi @leander-dsouza,

Yes, this is something I am aware of. There is #285 open for exactly this. I will try to update the URDFs this week and close the issue once it is merged if it is good with you.

Thank you,
Felix

leander-dsouza, 2023-04-03:
I can correct the limits for both the arms and create a PR soon to fix this issue.

felixmaisonneuve, 2023-04-03:
Sure, it will be greatly appreciated

leander-dsouza, 2023-04-03:
I have created the PR to address this issue - #294.

felixmaisonneuve, 2023-04-06:
Closing this issue since the pull request has been merged

Thank You again,
Felix

closed 2023-04-06 by felixmaisonneuve