Kinovarobotics/ros_kortex #307: How to perform motion planning with 2 robot arms using MoveIt

tanrb, 2023-08-31:
[image]

I have launched the kortex_dual_driver.launch file and able to see the real time position of the robot model on rviz. However I would like to simulate movements for each arm like the example_move_it_trajectories.py in motion planning on rviz. Any advice on how I can do that? Thanks for your help!!

yoraish, 2024-07-07:
@tanrb Did you end up figuring it out?

aalmrad, 2024-08-01:
Hello @tanrb ,

Please refer to the pull request: https://github.com/Kinovarobotics/ros_kortex/pull/331 to fix your issue. Please do not hesitate to reach out for further questions

aalmrad, 2024-10-16:
Closing because of inactivity. Please reply to this thread to reopen.

Best,
Abed

closed 2024-10-16 by aalmrad