Kinovarobotics/ros_kortex #317: Failed to use inverse kinematics service

ashuRMS, 2024-02-12:
## Description

One paragraph to explain the issue

## Version

ROS distribution : 

Branch and commit you are using : 

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

martinleroux, 2024-02-12:
The IK function from the API calls the same numerical algorithm as the robot does during runtime. 
The algorithm does not look for a final solution right away but instead, every millisecond (or as fast as possible when not used while the robot is moving), the robot identifies a unit vector in the direction of the final destination and takes a single step along that unit vector such that:

dq = pinv(J)*dr

with dq = displacement of the joint
J = jacobian
dr = a step that the robot expects to be able to make in 1 ms.

While this algorithm is very flexible with the type  of geometry it can support, it has the downside of being very sensitive to initial guess values because it cannot know whether or not it can reach the destination before it either gets there or runs into an issue.

Since you are hitting a joint limit, this means that the algorithm is virtually moving the robot in a direction which indeed hits a joint limit. Unfortunately, my only suggestion is to try even more guess values.

ashuRMS, 2024-02-12:
Thank you @martinleroux I thought that if I specified the guess as the home position, I would be able to get the accurate solution, but it is not working. Anyway, I will look for some other approach.

closed 2024-09-11 by martinleroux