Kinovarobotics/ros_kortex #323: how to control gen3_lite_2f gripper via ros_control

yuweiDu, 2024-05-30:
I use ros_control for gen3_lite control. The hardware interface is used as [](https://github.com/empriselab/kortex_hardware). The arm joints work well but something goes wrong with the gripper. I always get the error as:
```
runtime error: timeout detected: BaseClient::SendGripperCommand
Error Receiving: recvfrom() failed with error code : 104
terminate called after throwing an instance of 'std::runtime_error'
 what():  timeout detected: BaseCyclicClient::RefreshFeedback
/kortex_hardware died from signal 6
/kortex_hardware left a core dump
 Determined pattern '/tmp/rosmon-node-myw5SJ/core'
 Could not find a matching core file :-(
```
It seems that the error is related to `BaseClient::SendGripperCommand`. My OS is Ubuntu 20.04 and ROS is Noetic. 

Thanks in advance for any help.

smoya23, 2024-05-30:
Hi @yuweiDu ,

Could you provide more information on what command you ran ? 

> Error Receiving: recvfrom() failed with error code : 104

From the error messages, it looks like a comm issue, but I would like to try and reproduce this on my side. Are you able to reach the arm through the WebApp ? If so, can you open/close the gripper from there ?

yuweiDu, 2024-06-03:
Hi @smoya23 ,
Yes, of course. There is something missing in the original issue I wrote. I used the hardware interface provide by @madan96 (https://github.com/empriselab/kortex_hardware) and change the "finger_joint" to "right_finger_bottom_joint" in https://github.com/empriselab/kortex_hardware/blob/4a73db42c2b23f9855051fa3c327afe67a7950be/src/Gen3Robot.cpp#L171
```
  hardware_interface::JointHandle grp_vel_handle(
      jnt_state_interface.getHandle("finger_joint"),
      &cmd_vel[num_full_dof - 1]);
  jnt_vel_interface.registerHandle(grp_vel_handle);

  hardware_interface::JointHandle grp_pos_handle(
      jnt_state_interface.getHandle("finger_joint"),
      &cmd_pos[num_full_dof - 1]);
  jnt_pos_interface.registerHandle(grp_pos_handle);

  hardware_interface::JointHandle grp_eff_handle(
      jnt_state_interface.getHandle("finger_joint"),
      &cmd_eff[num_full_dof - 1]);
  jnt_eff_interface.registerHandle(grp_eff_handle);
```

By the WebApp, I don' know where to control the gripper. But I can control the gripper by call the service /my_gen3_lite/base/send_gripper_command. I think the connection is good here. 

Thank you!

smoya23, 2024-06-03:
@yuweiDu Unfortunately, we don't offer support for code that we do not provide. I would suggest trying out our driver and if you're able to run it, then it means that something is wrong with the package you're using. Then you can compare the differences and adjust your custom code

aalmrad, 2024-10-15:
Closing due to inactivity - reply to the thread if you need the issue to be reopened

closed 2024-10-15 by aalmrad