Kinovarobotics/ros_kortex #325: send_gripper_command is interupted by /my_gen3/in/joint_velocity topic

AlfaKeNTAvR, 2024-06-10:
## Description

Kinova Gen3 (firmware 2.5.2) gripper stops any movement initiated by send_gripper_command service call if publishing to /my_gen3/in/joint_velocity topic. Depending on the publishing rate the gripper might start some motion: almost closes on rate 1Hz, twitches a bit on >10Hz.

## Version

ROS distribution : 

Branch and commit you are using : noetic-devel 085b694280a77eb7be14ff2af86958ba18b633cf

## Steps to reproduce

```
roslaunch kortex_driver kortex_driver.launch ip_address:=192.168.1.12 start_rviz:=false start_moveit:=false gripper:=robotiq_2f_85 robot_name:="my_gen3"

rostopic pub -r 10 my_gen3/in/joint_velocity kortex_driver/Base_JointSpeeds "joint_speeds:
- joint_identifier: 0
  value: 0.0
  duration: 0
duration: 0" 

rosservice call /my_gen3/base/send_gripper_command "input:
  mode: 3
  gripper:
    finger:
    - finger_identifier: 0
      value: 0.0
  duration: 0"

rosservice call /my_gen3/base/send_gripper_command "input:
  mode: 3
  gripper:
    finger:
    - finger_identifier: 0
      value: 1.0
  duration: 0" 
```

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

On another Kinova Gen3 (older) which runs 2.2.0 firmware the behavior is as expected. Publish to the topic does not have any effect on the gripper service call. I have tested with both arms, same ros_kortex, just changed the IP in the terminal.

## Any other information

Kinova Gen3 (older) firmware 2.2.0 - works as expected.
Kinova Gen3 (purchased Summer 2023) firmware 2.4.0 - 2.5.2 does not work.

Maybe it is related to the ros_kortex version? It seems to work on my colleagues PC, I think he uses older ros_kortex version, but then it shouldn't work with the older Kinova too in my system.

AlfaKeNTAvR, 2024-06-10:
UPD1: Downgrading the firmware to 2.3.0 solved the issue, so the problem lies somewhere in later updates.

martinleroux, 2024-06-14:
Hi @AlfaKeNTAvR 

We will test this and report the bug to our engineering department if we can replicate it. However, looking at the source code of firmware 2.5.2, there are no changes in the communication layer, so in theory there should be no impact. Is there a change the act of updating your firmware solved the issue? If you re-updated to 2.5.2 does the issue reappear?

AlfaKeNTAvR, 2024-06-14:
Hey @martinleroux,

I don't believe the act of reuplading the firmware solved it, because the robot came with 2.4.0 originally and I have noticed this issue, so I updated to 2.5.2, but the issue still existed, hence I moved to 2.3.0 which our other Kinova Gen3 is running and it doesn't have the issue and it was fixed. 

I can give a try and update the other Kinova from 2.3.0 to 2.5.2 to see if it might be the new arms' issue.

aalmrad, 2024-10-15:
Hello @AlfaKeNTAvR ,

I tried to repoduce the issue on my end using a Gen3 7DoF firmware: 2.5.2, ROS Noetic, Ubuntu 20.04.4 and the latest version of the noetic-devel branch. I was able to command the robot joints and call the gripper service to open it and close it with no problems regardless of the publication rate on the joint velocity topic. Please let me know if you are still facing any problem with your robot and we can try to find a solution.

Sincerely,
Abed

AlfaKeNTAvR, 2024-10-15:
Hey @aalmrad,

As far as I remember we stayed at 2.3.0 and I haven't tried it on two of our other Kinova Gen3 arms. I will give it a try in the upcoming months and reply here about the results.

Thanks,
Nikita