Kinovarobotics/ros_kortex #326: Issues with Kortex Dual Arm Driver

yoraish, 2024-07-08:
## Description

I am attempting to use MoveIt! to control multiple Kinova Gen3 (with Robotiq 2F-85) robots together. The two arms and a laptop are connected via ethernet cables to a router. Both arms can be controlled properly via `roslaunch kortex_driver kortex_driver.launch gripper:=robotiq_2f_85 ip_address:=<IP>`. As of now, I could not get both arms to work together via `roslaunch kortex_driver kortex_dual_driver.launch <params>`. Would love advice here.

## Version

ROS distribution: Noetic on Ubuntu 20.04

Branch and commit you are using:  noetic-devel at d57a287cdc85031ae351089b82c71cb8cace16be

## Steps to reproduce
1. Created a new ROS workspace directory and cloned the `ros_kortex` package into its `src` directory.
2. Install moveit via `sudo apt install ros-noetic-moveit`. (I re-ran the command to verify up-to-date binaries.)
3. `roslaunch kortex_driver kortex_dual_driver.launch left_ip_address:=192.168.1.22 right_ip_address:=192.168.1.33 left_gripper:=robotiq_2f_85 right_gripper:=robotiq_2f_85`. (Example IP addresses).
4. Observed output: the RViz window opens up properly but the pose of the robots is incorrect (they are pointing up, all angles are zero). Additionally, the MoveIt! motion planning pane is not responsive and this message pops up in the terminal `[ERROR] [1720462627.560042924]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?` [image]

A few things I have attempted:
1. In an attempt to at least have the simulated robots to reflect their real poses, I have modified the launch file `kortex_dual_driver.launch` to add a namespace around each of the `kortex_arm_driver` includes:
    ```    
    <include file="$(find kortex_driver)/launch/kortex_arm_driver.launch" ns="$(arg left_robot_name)"> 
    ```
    This helps, and now the `joint_state_publisher` creates a meaningful `/joint_states` topic and the actual poses are reflected in RViz. MoveIt is still not working.
[image]
2.  I was wondering if MoveIt could need a dedicated `moveit_config` for this setup, so I created one with `setup_assistant` and tried to launch its `move_group.launch` from within (and not) the `kortex_dual_arm_driver.launch` file. I tried a few different configurations and changes in this approach but was not able to make progress.
3. I tried the `bug-fix/dual-arm-melodic` branch in multiple configurations. I can provide more details on this if needed, but I figured it might be better to stick with the Noetic branch. 

Any help or advice would be greatly appreciated!

yoraish, 2024-07-08:
Excitingly, progress was eventually made. We added the namespaces as described above, included a new MoveIt! configuration package's `move_group` launchfile, and corrected the names of the controllers within the `simple_moveit_controllers.yaml` file. As an example, this is a corrected controller name:

```
  - name: my_left_arm/left_gen3_joint_trajectory_controller
    action_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    default: True
    joints:
      - left_joint_1
      - left_joint_2
      - left_joint_3
      - left_joint_4
      - left_joint_5
      - left_joint_6
  ```

:)

closed 2024-07-08 by yoraish

martinleroux, 2024-07-09:
Thanks for letting us know!
We'll see if we can improve our documentation accordingly on our next update.