Kinovarobotics/ros_kortex #327: catkin_make issue

alieabd, 2024-07-11:
### Issue Summary

I encountered multiple issues while trying to build the `ros_kortex` package. Below are the steps I followed and the errors I encountered. Attached is a screenshot of the terminal with the error messages.

### Steps to Reproduce
I followed the steps provided in the [Github page](https://github.com/Kinovarobotics/ros_kortex), however I did't know how to use the [Download links](https://github.com/Kinovarobotics/ros_kortex#download-links) section:
1. **Install Conan and Dependencies**:
   ```
   sudo apt install python3 python3-pip
   sudo python3 -m pip install conan==1.59
   conan config set general.revisions_enabled=1
   conan profile new default --detect > /dev/null
   conan profile update settings.compiler.libcxx=libstdc++11 default
   ```

2. **Build the Workspace**:
   ```
   mkdir -p ~/catkin_workspace/src
   cd ~/catkin_workspace/src
   git clone -b noetic-devel https://github.com/Kinovarobotics/ros_kortex.git
   cd ../
   rosdep install --from-paths src --ignore-src -y
   catkin_make
   ```
### Errors Encountered
[image]

### Additional Information

- **Operating System**: Ubuntu 20.04
- **ROS Version**: Noetic
- **Conan Version**: 1.59

### Other information:
   you can see the whole log below:
```
Base path: /home/ali/catkin_workspace
Source space: /home/ali/catkin_workspace/src
Build space: /home/ali/catkin_workspace/build
Devel space: /home/ali/catkin_workspace/devel
Install space: /home/ali/catkin_workspace/install
####
#### Running command: "cmake /home/ali/catkin_workspace/src -DCATKIN_DEVEL_PREFIX=/home/ali/catkin_workspace/devel -DCMAKE_INSTALL_PREFIX=/home/ali/catkin_workspace/install -G Unix Makefiles" in "/home/ali/catkin_workspace/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ali/catkin_workspace/devel
-- Using CMAKE_PREFIX_PATH: /home/ali/catkin_workspace/devel;/opt/ros/noetic
-- This workspace overlays: /home/ali/catkin_workspace/devel;/opt/ros/noetic

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ali/catkin_workspace/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10")
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - kortex_move_it_config (metapackage)
-- ~~  - kortex_control
-- ~~  - kortex_description
-- ~~  - kortex_gazebo
-- ~~  - gazebo_version_helpers
[... 277 lines cut ...]
os_build=Linux
kortex_api_cpp:arch=armv7
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=6.4
kortex_api_cpp:compiler.libcxx=libstdc++11
kortex_api_cpp:build_type=Release
[options]
[build_requires]
[env]
TARGET=imx6
kortex_api_cpp/2.6.0-r.3@kortex/stable: Not found in local cache, looking in remotes...
kortex_api_cpp/2.6.0-r.3@kortex/stable: Trying with 'conancenter'...
ERROR: Failed requirement 'kortex_api_cpp/2.6.0-r.3@kortex/stable' from 'conanfile.py'
ERROR: Permission denied for user: 'None'. [Remote: conancenter]
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:99 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/home/ali/catkin_workspace/build/CMakeFiles/CMakeOutput.log".
See also "/home/ali/catkin_workspace/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

martinleroux, 2024-07-12:
It seems there is a problem in your Conan configuration which prevents it from finding Kortex API on our Artifactory server. In general, I advise wiping the Conan installation clean and starting over because it is easier than finding the issue, but in this particular case, there is a more convenient workaround. You can set the USE_CONAN option (line 7 in CMakeLists.txt to OFF and just fetch the API manually first.  Simply pick the version that corresponds to your environment in the list available on our artifactory server (https://artifactory.kinovaapps.com/ui/repos/tree/General/generic-public/kortex/API/2.6.0)

Hope this helps.

alieabd, 2024-07-13:
Thank you for your previous response; it resolved the initial issue. However, I’ve encountered another error. Below are the details:

**Error Output:**

[image]

You can see the output of catkin_make in the output.txt file:

[output.txt](https://github.com/user-attachments/files/16202709/output.txt)

**Observation:**
- The `kortex_API` directory is empty.

**Directory Structure:**
Here is the output of the `tree` command showing the directory structure:

[tree output.txt](https://github.com/user-attachments/files/16202710/tree.output.txt)

martinleroux, 2024-07-15:
The API package you downloaded from the Artifactory server should be added to the kortex_api folder.

closed 2024-09-11 by martinleroux