Kinovarobotics/ros_kortex #328: ROS1 and Docker Working -- Where to Put Info in the Repo?

peasant98, 2024-07-14:
## Summary

I have a working example of using Docker and ROS1 to control the Kinova Gen3 arm. I listed my steps here: https://reflective-blinker-149.notion.site/Kinova-with-ROS1-and-Docker-6e344ad97a204835a123a1cd7b0aa7cc. I was wondering where I could put this information in the repo -- with more detailed information in the repo.

## Use case

This tutorial will give users an easy way to control the Kinova Gen3 robot with Docker. The hope is that people who have newer versions of Ubuntu, Mac, and Windows could get this up and running.

## Alternatives

NA

## Possible issues

NA

## Additional context

NA

martinleroux, 2024-07-15:
Hi @peasant98 , thank you for your contribution.

If you intend to maintain this yourself, I would recommend creating your own repo, and then making a pull request adding a .md file in the 3rd Party folder of the ros_kortex repository to list it as a supported 3rd party tool (if you decide to go this way, let me know and I'll create the file in place and you will be able to simply add your link with a short description).

If you don't intend to maintain this yourself, you can create a Pull request after putting your content in a new subfolder of the 3rd party folder. It will need to go through our usual review process - I can already tell you that some rewording in your instructions would be required (while it is true that at the time of writing our ROS1 driver is more complete than for ROS2, we have plenty of improvements in our backlog that we are hoping to push in the next year).

aalmrad, 2024-10-15:
Closing due to inactivity - reply to the thread if you need the issue to be reopened.

closed 2024-10-15 by aalmrad