Kinovarobotics/ros_kortex #329: Robot semantic description not found, ros-noetic

Vasanth28897, 2024-07-15:
I am trying to implement the dual arm using this `kortex_dual_driver.launch` file from the `kortex_driver` package. This below command is used to launch 
```
roslaunch kortex_driver kortex_dual_driver.launch right_ip_address:=192.168.1.10 right_dof:=6 left_ip_address:=192.168.1.11 left_dof:=6 start_moveit:=true 
```
Getting this error after add the `moveit_ros_visualisation` topic in the rviz 
```
[ERROR] [1721050238.187388049]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
```
when i tried executing just kortex_driver.launch using this command 

```roslaunch kortex_driver kortex_driver.launch ip_address:=192.168.1.10 robot_name:=terminator dof:=6```

there are no problem or error shown in the terminal, and i can plan and execute the robot using the motion_planning in the rviz.

and i can see this parameter `/terminator/robot_description_semantic` in the rosparam list. And that robot_description_sematic located the srdf file in this location.

```This document was autogenerated by xacro from /home/pc/kinova_ws/src/ros_kortex/kortex_move_it_config/gen3_move_it_config/config/6dof/gen3.srdf.xacro\ ```

I added this below line in the `kortex_dual_driver.launch` file to locate the `gen3.srdf.xacro` file 
 ```   <param name = "robot_description_semantic" command="$(find xacro)/xacro --inorder $(find gen3_move_it_config)/config/$(arg left_dof)dof/gen3.srdf.xacro"/> ```

ended up the error like this
```
[ERROR] [1721049416.863728332]: Semantic description is not specified for the same robot as the URDF
[ERROR] [1721049416.863820634]: Joint 'joint_1' declared as part of group 'arm' is not known to the URDF
.
.
.
```

so what do i have to do, to implement the dual arm working using the `kortex_dual_driver.launch` file? This is urgent. Awaiting your response as soon as possible. 

Regards
K.S.Vasanth

martinleroux, 2024-07-16:
Hi @Vasanth28897 ,

We will be investigating the specifics of your issue as soon as we have time. In the meantime, have you looked at the bugfix/dual-arm-melodic branch published for issue #253 ? I understand you probably do not want to downgrade to Melodic, but you can use the code as a reference. Since you mention your issue is urgent, this is the best I can propose until we can replicate your issue.

aalmrad, 2024-08-01:
Hi @Vasanth28897 ,
Please refer to the pull request: https://github.com/Kinovarobotics/ros_kortex/pull/331
The added modifications will allow you to command two Gen3 robotic arms with no grippers and with the same dof (6 or 7).
Please do not hesitate to reach out for further questions

aalmrad, 2024-10-15:
This issue is closed due to inactivity. Please do not hesitate to reply to this thread to reopen