Kinovarobotics/ros_kortex #330: rosdep install error

Hal-Zhaodong-Yang, 2024-07-19:
## Description

Hello, when I am trying to install the packages running `rosdep install --from-paths src --ignore-src -y --rosdistro noetic`, it gives me unmet dependency error. And if I run `rosdep install --from-paths src --ignore-src -y`, ROS DISTRO not set error is raised.

## Version

ROS distribution : noetic
system: ubuntu 20.04

Branch and commit you are using :  noetic-devel

## Steps to reproduce
Here is the error message:
$ rosdep install --from-paths src --ignore-src -y --rosdistro noetic
executing command [sudo -H apt-get install -y ros-noetic-gazebo-ros]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-noetic-gazebo-ros : Depends: ros-noetic-gazebo-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-noetic-gazebo-ros] failed

And if I try to install ros-noetic-gazebo-dev, it complains about libgazebo11-dev not installed, which has more unmet dependencies.

aalmrad, 2024-09-11:
Hello,

Thank you for reaching out. First, I would like to point out that you should be able to run the following command without any problem `rosdep install --from-paths src --ignore-src -y` so please make sure that your ROS Noetic installation is properly sourced.
Second, it seems to me that Gazebo 11 is not installed on your system so please consult the following [link](https://classic.gazebosim.org/tutorials?tut=install_ubuntu) and make sure to read the instructions carefully before proceeding with the installation. Pleas let me know how things unfold.

Best,
Abed

aalmrad, 2024-10-15:
This issue is closed due to inactivity. Please feel free to reply to this thread to reopen.

closed 2024-10-15 by aalmrad