Kinovarobotics/ros_kortex #332: Gripper joint positions/velocities

Victeldo, 2024-08-08:
I'm working on a sim-to-real project where I need to pass in the Kinova Gen 3's sensor data, which includes the gripper sensor data (Robot IQ 2f 85). I need to get the positions/velocities of each of the 6 joints of the gripper. However, via the 
```
rospy.wait_for_message("/" + self.robot_name + "/joint_states", JointState)
``` 
I'm only able to get just one position/velocity value, which I assume just represents the aperture size/speed of opening/closing. Is there a way to get what I need for my project?

martinleroux, 2024-09-11:
Hi @Victeldo ,
The Gen3 only reports the position and velocity that you have found (it does represent the open/close state of the fingers, in %).
There is no way to obtain the data you are looking for. I even think the gripper itself does not report all of these values.

aalmrad, 2024-10-15:
This issue is closed due to inactivity, please feel free to reply to this thread to reopen.

closed 2024-10-15 by aalmrad