Kinovarobotics/ros_kortex #337: Latency in kortex_driver

fukui-masashi-k, 2025-02-20:
I am using kinova gen3 (6dof).

I am using moveit's compute_cartesian_path function to generate a plan and play it back at a specific timing.

I am concerned about the wait time between executing the trajectory and the arm starting to move.

I deleted the m_base->ValidateWaypointList() function in the source code below to eliminate the wait time, but I still feel a little wait time.
https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp

When I checked other wait times, it took about 1.0 to 2.0 seconds from when State changed from IDLE to PRE_PROCESSING_PENDING was called until action_notif_callback was executed again.

I would like to reduce this wait time. Do you have any good ideas?

best regards.

martinleroux, 2025-02-20:
Hi @fukui-masashi-k ,

How large is your trajectory? 

ValidateWaypointList() can indeed be removed - would you estimate that it removed roughly 50% of the wait time? The thing is, this validation is also run when ExecuteWaypointList() is called (this cannot be avoided). The point of having a separate validation function was to allow validation to be run "off-line" (in quotes here because you still need to be connected to a robot).

1-2s is larger than usual, but if your trajectory contains >1000 points, it's very plausible.  The only alternative to reduce this wait time would be to cut your trajectory in multiple segments, but it's unclear if the time saved on the more efficient processing will be cancelled out by the full stop required in the motion to end a list.

aalmrad, 2025-02-20:
Hello @fukui-masashi-k,

Just to add to Martin suggestions: You can also try to reduce the 1 sec sleeping time before sending the trajectory to the robot [here](https://github.com/Kinovarobotics/ros_kortex/blob/bf71d8ecdec874a88d02b1a2481784ae2a7fdba4/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp#L202):

fukui-masashi-k, 2025-02-21:
Hi! @martinleroux @aalmrad 
Thank you for your reply.

>How large is your trajectory?
The length of the plan was 181 points.

>ValidateWaypointList() can indeed be removed - would you estimate that it removed roughly 50% of the wait time? 
>Just to add to Martin suggestions: You can also try to reduce the 1 sec sleeping time before sending the trajectory to the robot [here](https://github.com/Kinovarobotics/ros_kortex/blob/bf71d8ecdec874a88d02b1a2481784ae2a7fdba4/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp#L202)
I change the sleep value(1000⇒100) and check wait time.
If ValidateWaypointList() is included, the time from "New goal received" to "ACTION_START" is 1.89sec.
If ValidateWaypointList() is not included, the time from "New goal received" to "ACTION_START" is 1.74sec.
Do you think it will be possible to complete the process in under 1.0 second just by modifying the source code in ros_kortex?

> 1-2s is larger than usual, but if your trajectory contains >1000 points, it's very plausible. The only alternative to reduce this wait time >would be to cut your trajectory in multiple segments, but [...]
I want the arm to move continuously without stopping, so I would like to avoid splitting the trajectory.

martinleroux, 2025-02-21:
Can you inject your trajectory in one of these examples (depending on whether your trajectory is angular or cartesian) and time it?
(Python) https://github.com/Kinovarobotics/Kinova-kortex2_Gen3_G3L/tree/master/api_python/examples/110-Waypoints
(C++) https://github.com/Kinovarobotics/Kinova-kortex2_Gen3_G3L/tree/master/api_cpp/examples/110-Waypoints

If the delay remains > 1sec, then there is nothing that can be done - it is due to internal processing of the trajectory. If not, it will mean that there is something in the ROS driver causing the delay.

fukui-masashi-k, 2025-02-28:
Thanks reply.
There is still a delay, so there is nothing more that can be done.
If there are no further discussions, it's okay to close this issue.

closed 2025-02-28 by martinleroux