Kinovarobotics/ros_kortex #344: Gen3 7DOF URDF incorrect meshes

justanothercoder99, 2025-06-16:
## Description

The mesh file location and names referenced in the Gen3 7DOF arm URDF are incorrect so importing this URDF throws a "RuntimeError: Accessed invalid null prim" on Isaac Sim. This error is not specific to Isaac sim since the URDF file that the simulation environments import is buggy.

## Version

ROS distribution : Any

Branch and commit you are using : branch - noetic-devel

## Steps to reproduce

1. Import the Arm Gen3 7DOF URDF file called `GEN3_URDF_V12.urdf`

## Expected behavior

The URDF should load the arm model in the simulation environment.

## Any other information

Any other information you believe the maintainers need to know.

justanothercoder99, 2025-06-16:
`
<robot name="GEN3_URDF_V12" version="1.0">
  <link name="base_link">
    <inertial>
      <origin xyz="-0.000648 -0.000166 0.084487" rpy="0 0 0" />
      <mass value="1.697" />
      <inertia ixx="0.004622" ixy="9E-06" ixz="6E-05" iyy="0.004495" iyz="9E-06" izz="0.002079" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kortex_description/arms/gen3/7dof/meshes/base_link.STL" />
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kortex_description/arms/gen3/7dof/meshes/base_link.STL" />
      </geometry>
    </collision>
  </link>
  <link name="Shoulder_Link">
    <inertial>
      <origin xyz="-2.3E-05 -0.010364 -0.07336" rpy="0 0 0" />
      <mass value="1.3773" />
      <inertia ixx="0.00457" ixy="1E-06" ixz="2E-06" iyy="0.004831" iyz="0.000448" izz="0.001409" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kortex_description/arms/gen3/7dof/meshes/shoulder_link.STL" />
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kortex_description/arms/gen3/7dof/meshes/shoulder_link.STL" />
      </geometry>
    </collision>
  </link>
  <joint name="Actuator1" type="continuous">
    <origin xyz="0 0 0.15643" rpy="3.1416 2.7629E-18 -4.9305E-36" />
    <parent link="base_link" />
    <child link="Shoulder_Link" />
    <axis xyz="0 0 1" />
    <limit effort="39" velocity="0.8727" />
  </joint>
  <link name="HalfArm1_Link">
    <inertial>
      <origin xyz="-4.4E-05 -0.09958 -0.013278" rpy="0 0 0" />
      <mass value="1.1636" />
      <inertia ixx="0.011088" ixy="5E-06" ixz="0" iyy="0.001072" iyz="-0.000691" izz="0.011255" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kortex_description/arms/gen3/7dof/meshes/half_arm_1_link.STL" />
      </geometry>
      <material name="">
        <color rgba="0.75294 0.75294 0.75294 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
[... 141 lines cut ...]
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://kortex_description/arms/gen3/7dof/meshes/bracelet_no_vision_link.STL" />
      </geometry>
    </collision>
  </link>
  <joint name="Actuator7" type="continuous">
    <origin xyz="0 -0.10593 -0.00017505" rpy="-1.5708 -5.5511E-17 9.6396E-17" />
    <parent link="SphericalWrist2_Link" />
    <child link="Bracelet_Link" />
    <axis xyz="0 0 1" />
    <limit effort="9" velocity="0.8727" />
  </joint>
  <link name="EndEffector_Link" />
  <joint name="EndEffector" type="fixed">
    <origin xyz="0 0 -0.0615250000000001" rpy="3.14159265358979 1.09937075168372E-32 0" />
    <parent link="Bracelet_Link" />
    <child link="EndEffector_Link" />
    <axis xyz="0 0 0" />
  </joint>
</robot>
`

justanothercoder99, 2025-06-16:
Seem this is has been fixed in a newer branch `noetic-devel` but the kinova website directs me to the `kinetic-devel` branch which has the bug. It would be good if you fixed the bug in both the branches. I have a branch ready for the fix for `kinetic-devel` if you would like me to push that and make a pull request

aalmrad, 2025-07-15:
@justanothercoder99 thanks for pointing this out. Please do not hesitate to open a pull request so we can review and potentially merge the modifications. Thanks in advance for your contribution.

justanothercoder99, 2025-07-15:
@aalmrad 
PR for the same:
https://github.com/Kinovarobotics/ros_kortex/pull/345/files

closed 2025-07-16 by aalmrad